pbatard/rufus #1103: rufus bricked 2 brand new usbs

tuggerfub, 2018-03-30:
quit blaming us users 

there is something wrong with newer iterations of rufus 

and it is on the development end not ours 

I had formatted and checked both for bad sectors prior to using rufus
no problems 

use rufus

sector check takes an hour
sector check finds every sector bad

now my things are bricked 

when are the developers gonna be held accountable smh

pbatard, 2018-03-30:
> when are the developers gonna be held accountable smh

When either we can replicate the alleged issues (which, in months and months of using Rufus on pretty much a daily basis against a variety of flash drives we have not been able to do) or we receive credible evidence (i.e. one with a USB trace that shows that there was a command susceptible to damage a drive) from either a flash drive manufacturer (who, once more, should have every inclination in the world to demonstrate to us that Rufus has a software flaw that damages drives, if the end result is that they have to spend money RMA'ing drives) or someone with enough technical expertise, such as the many power users who also use Rufus on a daily basis, to provide the kind of technical evidence that is required to demonstrate the alleged issue.

Sorry but, as long as no technical expert seems to be in a position to replicate the issue, then the burden of evidence is on __you__. Else, you are simply dropping a baseless accusation, and implying that we "know" that there is an issue where Rufus can damage drives (while we have repeatedly provided technical information as to why this is logically impossible), and are somehow sitting on a mountain of bricked flash drives just to spite users.

As much as I can understand your frustration, as well as trying to place the blame on something, the reality does paint a very different story than what you are implying. For instance, are you aware that there does exist USB Flash Drives with problematic firmwares that will result in bricked drives under conditions that have nothing to do with the application being used? Or that it is very possible that hardware environment can have an effect on a flash drive? Or that, depending on how they were manufactured, a batch of flash drives may very well fail in the same fashion (for instance due to a falty batch of flash memory cells)?

Moreover, if a bad blocks check does find bad sectors, then your flash drive has issues, period. The fact that it is slow is irrelevant (of course checking every individual block __is__ going to be slow!).

I know you want to blame someone, so I assume that this statement is not going to sit well, but please try to apply logic rather than trying to blame the first thing or person you can just because you are angry that your drives have failed.

Again, __if__ there is an actual issue, it should be __exceedingly__ easy to demonstrate by technical experts. Logically then, if after many years of people complaining that they believe that Rufus has damaged their drive (but in no larger number than people I can see on Stack Overflow complaining that their flash drives have suddenly become read-only or damaged __outside of any use of Rufus__), this hasn't happened, it must mean that, despite what a handful of users may think, Rufus is actually safe to use.

tuggerfub, 2018-03-30:
the reality is if you break people's stuff you get taken to court 
and that's where you will wind up. 

all you say in response to your destruction: 

"software can't damage hardware" :  this is patently false and anyone who has ever managed hardware knows this 

"bla bla must have been some other process" : No, it's RUFUS. This is why you have a gigantic defensive post about it and why you author gigantic defensive posts about it with copypasted excuses. 

"it hasn't been reproduced" : because you put your head in the sand when people tell you YOUR SOFTWARE makes THEIR USB DRIVES unusable (repeatedly!)

all you do is pretend that you're not at fault because your huge ego sniffs your monthly download count like it means something. It doesnt! some journos shared your software when it worked properly. 

guess what? They'll bury it when the realize it's not

pbatard, 2018-03-30:
Then I will strongly encourage you to read the license, with is displayed in the application, and which is what will hold in a court of law.

Also, I haven't been making generalisations (like "software can't damage hardware"), but instead I have explained why in the specific case of the operations carried out by Rufus, this can simply not be the case without the existence of a Windows bug.

Now, if all you want to do is reply in anger, rather than hear logic, I will have no choice but to take the appropriate measures.

tuggerfub, 2018-03-30:
You are breaking people's property and hiding behind it with excuses, and now that you are confronted with the reality that your one claim-to-fame utility program isn't all it's supposed to be you hide behind a license agreement. Whatever little piddle you write in your licenses doesn't circumvent laws, and one of us seems to know a little bit more about property law than the other (and is in the process of making arrangements with the Irish claims system). 

You can take whatever spooky measures you want, I care about consumer advocacy, and by being negligent and combative you're putting people who need a bootable USB utility to address other problems with their systems in even worse shape.

pbatard, 2018-03-30:
Again, if you want to accuse someone or something, then the burden of proof is on __you__. Please liaise with your flash drive manufacturers and prove your allegations.

tuggerfub, 2018-03-30:
As other users have told you: The evidence is right before you. Many users have taken the time to make these issues known, you have only taken the time to minimize and gloss over their claims. You act purely in bad faith and it shows.

closed 2018-03-30 by pbatard