pbatard/rufus #1104: [enhancement] Completed Dialog

cawoodm, 2018-03-31:
Noticed that when Rufus is finished it just plays a sound but shows no message/dialog that it completed successfully. When you're doing other things you can be unsure of which app caused the sound: was it Rufus? Was it an error? Is Rufus finished doing what I asked? Switching to Rufus you just see it sitting happily there:

[image]

Long story short: show a message/dialog that you're done Rufus!

cawoodm, 2018-03-31:
You have a status bar at the bottom. How about setting the text to "Done writing to USB!"?

pbatard, 2018-03-31:
READY and a completed status bar do the same.
For the record, displaying any new message on the UI requires updating more than 35 translations, which takes between 1-2 months, and I am already in the process of updating translations for Rufus 3.0. But even if that wasn't the case, the progress bar and a READY __are__ more than enough to tell you all the information you need. If you heard a bell (which was already added at user's request to indicate completion), got an icon flashing on the taskbar (another feature that was added at user's request to indicate completion), see a full progress bar and a READY status, you already have all you need to know that Rufus is complete.

And at this stage, because I've literally been adding features over and over to indicate completion, I have very little patience to add any more, because every 6 months or so, someone seems to want a completely different way to do so. So I will kindly ask: please learn to read all the hints you __already__ get. or if you still think that this is not enough, use another application.