pbatard/rufus #1108: cannot use UEFI:NTFS and Ubuntu 14.04.5 amd64 image

ggrellqd, 2018-04-12:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [x] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Dear development team of Rufus,
I have the following situation:
I need to create a bootable USB stick running the Ubuntu 14.04.5 LTS image (amd64). This stick needs to use your UEFI:NTFS feature, because I supply a customized filesystem.squashfs larger then 4gb afterwards, which violates the fat32 restriction.

However, when I select NTFS and no matter what partitioning (MBR for UEFI or GPT for UEFI), I end up with a stick that is bootable, but ends up at the grub command line. (Note: this is before I supply the larger filesystem.squashfs)
On the contrary, when I use Fat32, the stick is bootable. 

I was able to see that the last message that is printed when the UEFI drivers are loaded  (before the grub command line is shown) is:
"unknown filesystem" 
his is independent on the kind of stick I generated or the kind of machine I am trying to boot on

I checked the USB for bad blocks, tried different ports, etc.
Can you suggest a possible solution? It has to be UEFI bootable. Legacy (BIOS) is not an option anymore.

Here is the log from an attempt with a MBR+UEFI partitioning, wich also has the checking for bad blocks in it
Log
---
```
Rufus version: 2.18.1213
Windows version: Windows 10 64-bit (Build 16299)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0407
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found USB 2.0 device 'USB DISK 2.0 USB Device' (13FE:4200)
Using autorun.inf label for drive D: 'Install Ubuntu'
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1885, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {0FD35FFA-2E28-4FF4-A1A4-B10549537705}
Max parts: 128, Start Offset: 17408, Usable = 15510501888 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {F1473C8A-A02D-4763-8CB7-4F68E4E2377A}
  Size: 14.4 GB (15508394496 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {72BFBE87-6608-4C70-854B-38B8191EA78D}
  Size: 256 KB (262144 bytes)
  Start Sector: 30291881, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.05/20140113 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
[... 545 lines cut ...]
     22528 KB occupied by the log file.
  14025848 KB available on disk.
      4096 bytes in each allocation unit.
   3786229 total allocation units on disk.
   3506462 allocation units available on disk.
NTFS Fixup completed.

Found USB 2.0 device 'USB DISK 2.0 USB Device' (13FE:4200)
Using autorun.inf label for drive D: 'Install Ubuntu'
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1885, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0xD2842C16
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.4 GB (15508394496 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 256 KB (262144 bytes)
  Start Sector: 30291881, Boot: No

```

pbatard, 2018-04-12:
Thanks for the report.

Two things:

1. NTFS support for Linux boot is very much __experimental__ at this stage. The reason for this doesn't have to do with Rufus but purely with distro maintainers, as they need to make sure that their distro does indeed support boot and installation from NTFS, just like most of them make sure is supports boot and installation from FAT32 (which is __NOT__ a given, and always requires proactive maintainer support).

2. From what I can see, the Ubuntu maintainers did __NOT__ add the NTFS module to their GRUB EFI bootloader. Only `fat.fs` (for FAT32) appears to be present. Without the NTFS module then, boot from an NTFS partition will typically fail with the error you got.

As, again, this is not something Rufus can compensate for (the EFI GRUB bootloaders tend to be uniquely tailored for each distro, so it's not really possible for Rufus to try replace them with its own version that would include NTFS support), __you should log an issue with Ubuntu support__, to let them know that they should include the NTFS file system module in the `EFI\BOOT\grubx64.efi` GRUB bootloader they provide on their ISO. This is __VERY EASY FOR THEM TO DO__, and it will only add a few KB to their bootloader which is peanuts when considering that it'll ensure that users just like you are not being left stranded.

Oh, and in case you may want to ask why I shouldn't be the one creating the issue with Ubuntu, I'm just going to point out that I've been logging similar reports with other distros (e.g. [Manjaro](https://bugs.manjaro.org/#ticket/zoom/17934)), but I am seeing these getting mostly ignored, so I no longer think it is a valuable use of my time. Hopefully, the Ubuntu distro maintainers may be more receptive to actual users...

Finally, I should point out that I had a look at the latest LTS BETA (`ubuntu-18.04-beta2-desktop-amd64.iso`) and the issue of not including the NTFS file system module in the GRUB EFI bootloader still seems to be present...

ggrellqd, 2018-04-12:
Thank you for this insightful answer!
I found a workaround by putting the ntfs modules to the fat partition together with the UEFI:NTFS bootloaders of yours. Then I can load the module from the grub command prompt, switch to the ntfs drive and boot the image. Clumsy but it works.
I will still consider filing a report to the Ubuntu authorities as you suggested. I seems to be very easy to fix just as you said.

Thanks again for helping me out!

pbatard, 2018-04-13:
Okay. I'm going to close this issue then.

closed 2018-04-13 by pbatard

pbatard, 2018-04-13:
For reference, I have now logged a bug report with Ubuntu for this. See https://bugs.launchpad.net/ubuntu/+bug/1763738

ggrellqd, 2018-04-14:
Thank you for opening the issue. 
I think you found a more appropriate formulation then I would have.
Maybe they will fix it. Lets see.