pbatard/rufus #1114: Please update the first figure on this webpage

Kractal, 2018-04-24:
Please update the first figure on this webpage:https://rufus.akeo.ie/?locale=zh_CN.
I download rufus-2.18.exe,which SHA256 value is  E82ABD7F2C8F8C866141634A1CE10DA8EBF3C58B68CB2EAA351345777BB3F67C, and run it on Windows 1709 16299.402.
The original figure is:
[image],
Please update it to:
[image]

pbatard, 2018-04-24:
Done. I'm probably going to update all the pictures once Rufus 3.0 is out anyway, as the interface will be completely different.

closed 2018-04-24 by pbatard

Kractal, 2018-04-25:
Fine. I could provied the screenshot of the zh_CN version, please send me an email if you need my help.