pbatard/rufus #1117: Question: Grub2 Loading and Welcome message

Sporesirius, 2018-04-26:
Hello,
Is it possible to remove the Loading and Welcome message or to build an option?

pbatard, 2018-04-26:
I have no plans to do that, especially as it may come from the options the maintainers of the ISO you are using decided to go with rather than Rufus.
As a general rule, hiding information during early boot is a bad idea, because then people are left clueless if that process doesn't work as expected.

It would also help if you indicated why you think the message should be hidden? Is it causing an actual issue? Or is it more of a personal cosmetic annoyance (in which case I hope you can understand that it's difficult for an application to add features that boil down to individual taste)?

pbatard, 2018-05-01:
Now follow up &rarr; Closed.

closed 2018-05-01 by pbatard