pbatard/rufus #1123: Please add Kaspersky antivirus as incompatible to FAQ

F3000, 2018-05-05:
The title explains it. 

With kaspersky i see the error that the device is in use by another process

pbatard, 2018-05-05:
1. Can you please send a log when attempting to format a drive in Rufus while Kaspersky is running. Recent versions of Rufus try to identify conflicting processes, so I would very much like to see what the log reports.

2. Most of the problematic issues with antivirus software have to do with also enabling a specific drive access prevention policy, such as "Device Control" in the case of ESET or "Peripheral Control Policy" in the case of Sophos. I therefore strongly suspect that the same is true in the case of Kaspersky, and that you should be able to fine tune the policies to have Kaspersky running while also enabling low level access to flash drives. Can you please dig a bit further in your policy control and see if there's a specific option that is responsible for preventing drive access?

pbatard, 2018-05-07:
Any update on this? I __must__  see the log before I can do anything with this report, as I need to have some reliable data to corroborate that the issue really has to do with Kaspersky and not something else (which the log would show).

pbatard, 2018-05-08:
Again, the reason why I want to see the log is that Rufus has the ability to list the process that is preventing access to a drive when it can't access it. This means that with a log I should be able to confirm if the conflicting process is really Kaspersky or something else.

And again, most of the antivirus software listed in the incompatibility list have some level of control over the feature that prevents access to the drive, which can usually be disabled without turning off the antivirus.

I'm afraid I simply cannot add a blanket statement telling users that Kaspersky is incompatible with Rufus when:
1. Only __one__ user has reported the issue (whereas I expect many people to be affected by this, and therefore it is a bit puzzling that only one person would report it)
2. I haven't seen a log that may indicate which process causes the access issue
3. Kaspersky users have yet to indicate if, like the other incompatible antivirus software out there, they have the ability to enable/disable the specific feature that prevents access to USB drives without turning off Kaspersky altogether.

What I am asking for shouldn't be that difficult to provide from Kaspersky users. If they can't, then I'm going to have to suspect that maybe the issue is not as clear cut as OP reported, which is why I will hold off adding Kaspersky to the list of incompatible software.

F3000, 2018-05-08:
I will have a look at the log files and on thekaspereky setting. But I can use the stick in normal way.

pbatard, 2018-05-15:
I'm going to close this issue for now. Feel free to provide the data I requested at your convenience, so that I can decide whether I need to add Kaspersky to the list or not.