pbatard/rufus #1127: Extracting ISO from bootable USB drive

JonShard, 2018-05-10:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hello, thank you for your time in advance. I hope I am asking through the appropriate channel.

Is it possible at all to extract the ISO from the copied-to-disk content of a bootable usb drive?

I have a home exam tomorrow in computer graphics, my ubuntu installation encountered some kind of error causing it not to boot. Not even into recovery mode. I have this copy of the content of the usb drive i used to install ubuntu. But not the usb stick itself. I tried copying them to a newly formatted usb drive, however the bios said there is no bootable OS on it. 

There are many ISOs of ubuntu, and trying to install it, I could only find that one ISO that had support for both my touchpad controller, and my networking card. Trying them one by one would take a really long time. 

Is it possible to re-fit / extract then reinstall the ISO onto a usb drive? So I can format the Ubuntu partition?

JonShard, 2018-05-10:
Or maybe just finding a unique identifier for the ISO installed on it, that would make it possible to find the correct ISO on Ubuntu's pages?

pbatard, 2018-05-10:
No, I'm afraid that is not possible. There's a conversion process occurring between the ISO and the USB (some boot files need to be replaced), and I would need to add a lot of code and extra content to do the reverse operation... for very little benefit as almost everybody would either:
- Still have a copy of the ISO they used
- If not, be able to recall the ISO they used, and where to download it

At this stage, I expect your situation to be fairly unique, and, as you can understand, considering that my time is very limited, I see little point of investing a lot of it just to help the handful of people who may create a bootable USB, and then completely forget about the ISO they used to do that (which, to be fair, is their own problem).

As to adding an extra file (e.g. some kind of readme) to indicate the ISO used, I'm afraid I don't want to that, because the goal of Rufus is to convert ISOs with as little modifications as possible, so files are only added when necessary (and, in case people want to point to the `autorun.inf` that may be created by Rufus, this is actually a result of wanting to be able to display more than the 11 uppercase non-international characters that FAT32 restricts a volume label to, which I consider useful enough to warrant being added, rather than just something that gets added for the sake of it). I don't see adding a `readme.txt` for a couple of forgetful users as a good justification, sorry.

Besides, most Linux distros come as ISOHybrid these days, and if you write them in DD mode, then Rufus cannot add anything to it, so the suggestion would only be applicable half of the time.

My advice: Figure out the label of your USB drive, or look for a readme or version file. That should tell you the ISO you should use.

&rarr; Closed

closed 2018-05-10 by pbatard