pbatard/rufus #1128: Changelog: Mentioning

SetTrend, 2018-05-11:
In the [Changelog](https://github.com/pbatard/rufus/blob/master/ChangeLog.txt) it reads:

```
Version 3.0 (2018.??.??):
  UI redesign to follow the flow of user operations
  (with thanks to Fahad Al-Riyami for the concept)
```

I'm disappointed. Three years ago I already [suggested a new flow of the UI redesign to you](#807). I even implemented and provided all the sources and a working build at [SetTrend/rufus](/SetTrend/rufus).

And all that is not even worth mentioning me?

pbatard, 2018-05-11:
And, if you look at #807, you'll recall that I already told you at the time that this was a duplicate of #117.

I also told you explicitly that _"I already have a pretty clear idea of what I want to do (which is different from your proposal)"_

I didn't use any of the suggestion you made, except for the ones that may have __coincided__ with the ones I already had in mind, which, if that was the case, was purely accidental.

For the record, this is what Fahad submitted, which as you can see is exceedingly close to the new design:

[image]

I don't see much in your proposal that resembles the new UI (which is logical, since I have long discarded it). As a result, you don't get any credit since none of what you proposed was either used in the new design or original (i.e. something that we both came up with on our own).

closed 2018-05-11 by pbatard

pbatard, 2018-05-11:
For reference, in case you haven't seen it, this is the new pretty much final design for Rufus 3.0:

[image]