pbatard/rufus #1129: UEFI NTFS GPT Load failure: Unsupported

comnuoc, 2018-05-13:
Checklist
---------
- [v ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [v ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [v ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [v ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [v ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
Rufus version: 2.18.1213 (Portable)
Windows version: Windows 10 64-bit (Build 16299)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'SanDisk Ultra USB 3.0 USB Device' (0781:5591)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3738, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 28.6 GB (30752000000 bytes)
  Start Sector: 0, Boot: No
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Size: 4.4 GB (Projected)
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: Win10_1803_English_x64.iso (4.4 GB)

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'E:' as volume mountpoint
Requesting lock...
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Bad Blocks: Checking from block 0 to 60062499
Bad Blocks: Using offset 349 for fake device check
Bad Blocks: Writing test pattern 0xAA
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)
Clearing MBR/PBR/GPT structures...
[... 1024 lines cut ...]

Found USB 3.0 device 'SanDisk Ultra USB 3.0 USB Device' (0781:5591)
1 device found
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3738, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {F3B6504A-B006-4A5A-A855-88ECFEE5D157}
Max parts: 128, Start Offset: 17408, Usable = 30751965696 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {F596D363-74D9-4AE7-AE44-EF8084516F59}
  Size: 28.6 GB (30749838336 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {DA2D2701-47F0-4F8C-A444-1885A3C57A4B}
  Size: 256 KB (262144 bytes)
  Start Sector: 60060326, Attributes: 0x0000000000000000

```
[image]

[image]

pbatard, 2018-05-13:
Your `install.wim` is less than 4 GB. What happens when you try to create the same USB in FAT32 mode. You certainly don't seem to use UEFI:NTFS to boot that ISO. Furthermore, as per the UEFI:NTFS screenshot, your UEFI firmware already has an NTFS driver, so you shouldn't have had to use UEFI:NTFS to boot your NTFS partition, as the firmware should have proposed you to boot directly from it.

Also, I'd like to have the SHA-1 of the ISO you used to confirm that it's an official one.

All in all, I kind of get the feeling that there are still some details missing from your report, because what you are attempting to do is a bit illogical, so there must be a reason why you're trying to do it that way... Did you alter the content of the USB in any way after it was created? I may have to ask you to upload your `efi\boot\bootx64.efi` somewhere so that I can check it out because if I find that I don't get the same issue on the UEFI systems I have whereas you do, then the only logical conclusion is that this is an issue with your UEFI firmware. have you tried to boot the same USB on a different UEFI machine? If so, did you get the same error?

comnuoc, 2018-05-13:
Hi @pbatard ,

I'm using HP Zbook 15 G3, bios: N81 ver. 01.25 03/29/2018.
I'm using dual boot Windows 10 v1709 and Linux Mint 18.3.
I created usb as the following:
- Boot to Linux Mint.
- Download iso from https://www.microsoft.com/en-us/software-download/windows10.
- Boot to Windows 10.
- Create usb via Rufus.

I didn't change anything after usb was created.

Iso sha1: efe24d454093cacdf2905014d9852f648fe59722
Efi from UEFI_NTFS partition: 
[bootx64.efi.tar.gz](https://github.com/pbatard/rufus/files/1998461/bootx64.efi.tar.gz)
Efi from Iso data partition:
[bootx64.efi.tar.gz](https://github.com/pbatard/rufus/files/1998464/bootx64.efi.tar.gz)

pbatard, 2018-05-13:
Thank you very much for this.

I can replicate the issue with your `bootx64.efi` (the one from the ISO partition). But if I use the `bootx64.efi` from an official Microsoft ISO, such as `en_windows_10_consumer_editions_version_1803_updated_march_2018_x64_dvd_12063379.iso`, I don't get the error.

More interestingly, your `bootx64.efi` also fails to boot (or at least behave in the manner expected from the regular Microsoft version) when used on a FAT32 partition, i.e. without any involvement of UEFI:NTFS or Rufus.

Did you create the ISO using the Media Creation Tool?

And again, can you please tell what happens if you use FAT32 instead of NTFS as the target file system (basically, since there is no file larger than 4GB on this ISO, you should use GPT __and__ FAT32).

pbatard, 2018-05-13:
Yup, the `bootx64.efi` from your ISO is definitely corrupted, hence your issue, and it has nothing to do with Rufus.

An x64 EFI executable should start like with `MZ` and contain a PE/COFF headers. Yours has none of that, which means it is __NOT__ a working EFI application.

In an hex editor, the beginning of a working EFI executable should look close to this:

[image]

You may want to use an hex editor with the `efi\boot\bootx64.efi` from your ISO to find out how the actual file on your ISO looks, and whether it starts with `MZ`. If it does, then I suspect that your USB is starting to fail and is corrupting data. If not, then you will need to recreate your ISO again, as the one you have is corrupted.

comnuoc, 2018-05-13:
Yes, my iso file is corrupted. I have just downloaded the new Iso file, and it's working.
It's my mistake, sorry.
Thank so much.

closed 2018-05-13 by comnuoc