pbatard/rufus #1132: Tooltips in RTL UI are actually LTR'd

ItielMaN, 2018-05-14:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------
Tooltip shown in RTL UI of Rufus are infact LTR.
In #868 there's:
> I think our RTL tooltips are broken — Why the £$^& are they not right justified?

... which is checked V, but in Rufus 3.0.1278 beta the tooltips are still LTR'd.

Log
---
Not needed, but as you insisted:
```
Rufus version: 3.0.1278 (Beta) (Portable)
Windows version: Windows 10 32 bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x040D
Will use default UI locale 0x040D
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'he-IL'
NOTE: The Hebrew (עברית) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Disk size: ‏8‏GB, Sector size: 512 bytes
Cylinders: 973, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0FBBB283
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8003256320 bytes)
  Start Sector: 2048, Boot: Yes

```

On a side note, I'm starting now to improve the Hebrew language part of Rufus, and fill in what's missing.

pbatard, 2018-05-14:
Do you have a screenshot to share? Or at least can you point to a sequence of operations to do to see the issue?

For instance this is what I see on my system when hovering over the `?` symbol left of _BIOS (or UEFI-CSM)_ when FreeDOS is selected:

[image]

This text used to be left-justified and is now showing right-justified after I applied the fix. So I'd really like to have a screenshot of what you see, and where exactly you see it, as I have no idea what I should fix!

ItielMaN, 2018-05-14:
I don't do anything special really. Just launch Rufus and hover some items.
[image]
Here, the tooltip translates to "That allows boot selection and can masquerade the MBR Install BIOS USB driver ID", if read from right to left.
The same unreadability applies to "Target system" and others, but for some reason not to the V button that calculates SHA and all that stuff.

pbatard, 2018-05-14:
Well, that tooltip looks right aligned and therefore what I'd expect for RTL languages...

Aren't RTL languages supposed to be right-justified? It kind of makes sense to me that if you're reading from Right to Left, then the text should be aligned right where possible. Is that not the case??

pbatard, 2018-05-14:
Please be mindful that, as I pointed out, there's a big different with left or right alignment of text, and whether the text that is aligned makes sense due to presence or absence of Unicode ordering marks, which is a different issue.

ItielMaN, 2018-05-14:
> Please be mindful that, as I pointed out, there's a big different with left or right alignment of text, and whether the text that is aligned makes sense due to presence or absence of Unicode ordering marks, which is a different issue.

Oh. So.. yeah, I meant the markers (RLM and such).
Usually apps fix it in code, but I have no problem appending RLMs to every string, to fix this manually.
Or you'd rather I wouldn't do that, and that you'll fix it in later versions?

pbatard, 2018-05-14:
Okay, then we're on the same page.

> Usually apps fix it in code

Do you have examples of that? Because, even when trying to do it manually, I haven't found a good logic on where to insert the marks, let alone which ones, so that Windows will display mixed text as I want. This means that I'll be hard pressed to code something that does it automatically.

I was basically planning to fix that __manually__ in 3.1 (by inserting marks), when I have a chance to look at it more closely, because this will definitely take time, and it's not something I want to do between BETA and release. On the other hand, if you can figure out where to insert marks in your updated translation, so that the text displays as expected, that would be great help.

pbatard, 2018-05-14:
For the record, this issue is the _Try to improve RTL switching in UEFI/CSM messages_ point in #1095 (issues to fix in 3.1), which was moved from the already long list of stuff to do in #868 (3.0 issues) with the note: _ I'll leave that as a job to __tomorrow me__! (i.e. moved to 3.1)_

In other words, this is something I am already aware of, but that I'm not planning to fix by in 3.0.

pbatard, 2018-05-14:
Btw, if you are going to work on the Hebrew translation update (much appreciated), can you please e-mail me at: pete@akeo.ie so that I can put you in touch with the current Hebrew translator (who chose not to disclose his e-mail in the About dialog). He hasn't sent an updated translation yet, but that doesn't mean he's not working on it, and I'd hate to have one of you find out that you are duplicating the other's effort.

At the very least, I want to send an e-mail to the both of you indicating that you are working on the update, so that he can chime in if needed.

ItielMaN, 2018-05-14:
> Do you have examples of that?

Poedit maybe? CDBurnerXP? I have personally translated both of them.
I can't think of more for now but OTOH I can't remember any **other** apps that have issues with RTL tooltips. So this must mean something.
Anyway I'll think of some more.

> I haven't found a good logic on where to insert the marks, let alone which ones

I don't think this should be hard to implement (though maybe not efficient, and better alternatives may exist). Simply append RLM to every string on RTL languages. Worst case scenario, untranslated strings will have the dot at the beginning of the string (".Example").

> Btw, if you are going to work on the Hebrew translation update (much appreciated), can you please e-mail me at: pete@akeo.ie so that I can put you in touch with the current Hebrew translator

Will do.

ItielMaN, 2018-05-14:
> The same unreadability applies to "Target system" and others, **but for some reason not to the V button that calculates SHA and all that stuff**.

Any idea why this happens? It may be a key for fixing the rest of the tooltips.
[image]
Looks perfectly normal, and there are no markers in the translation for this string.

pbatard, 2018-05-15:
> Any idea why this happens?

Single line vs multiple lines, or, most likely, there are no parenthesis or dashes with LTR terms.

The problem with this issue is that we need to __fight__ Windows' automatic "smart" switching between RTL and LTR when it detects stuff like English terms, parenthesis, hyphens and line breaks, and this is where our problem stems from. From what I could see, Windows seems to have a mind of its own on where LTR terms should be placed because Microsoft added some custom logic for that... which doesn't always work as expected. So, as far as I am aware, we need to guess that broken logic and fight it back.

pbatard, 2018-05-17:
With Itiel's help, it seems we have figured out a solution for this.

The issue was that when you apply the `WS_EX_LAYOUTRTL` extended style to `CreateWindowEx()` when creating the tooltip (which is the __only__ way you can get right alignment for the tooltips), Windows somehow switches the layout to LTR for multiline tooltips (which we handle through `WM_NOTIFY` &rarr; `TTN_GETDISPINFO`).

At this stage, you would think that the solution, as mentioned [here](https://msdn.microsoft.com/en-us/library/windows/desktop/bb760258.aspx), would be to apply `TTF_RTLREADING` to the `NMTTDISPINFO`'s structure `uFlags`, because that would seem like the most logical thing to do to attempt to fix the problem... but that is where you are wrong. Logic and Windows programming have nothing in common!

What you need to do instead is __REMOVE__ the `TTF_RTLREADING` attribute from `uFlags` to tell Windows that you don't want to use RTL there. Once you have done that, then you get RTL. Yeah, that makes perfect sense: If you want Windows to give you something, you just need to tell the OS that you don't want it...

commit 4a219b36 (2018-05-17): [loc] fix various RTL localization issues

With thanks to Itiel Yossefian

* Fix a potential buffer overflow in lmprintf for RTL languages
* Automatically apply RLE/PDF to all RTL messages, and remove the RLE/PDFs from the .loc
* Fix Windows messing up of multiline RTL tooltips (The trick is, if you want actually
  want RTL, you need to *disable* RTL... Sure, Microsoft, that makes a lot of sense?!?)
* Also properly scale the length of the multiline tooltips according to the zoom factor
* Closes #1132
commit c3aaa0c2 (2018-05-17): [loc] fix various RTL localization issues

With thanks to Itiel

* Fix a potential buffer overflow in lmprintf for RTL languages
* Automatically apply RLE/PDF to all RTL messages, and remove the RLE/PDFs from the .loc
* Fix Windows messing up of multiline RTL tooltips (The trick is, if you want actually
  want RTL, you need to *disable* RTL... Sure, Microsoft, that makes a lot of sense?!?)
* Also properly scale the length of the multiline tooltips according to the zoom factor
* Closes #1132
closed 2018-05-17 by pbatard