pbatard/rufus #1133: Translate Ukraine

Alexander1414x, 2018-05-14:
Hello! Im Alexander , I want help to Ukraine translate .
Please give me file for language .

Thank you .
[image]

Alexander1414x, 2018-05-14:
For latest version 3.1

pbatard, 2018-05-14:
Please see [this guide](https://github.com/pbatard/rufus/wiki/Localization#Editing_an_existing_translation) that explains all.

There's no separate file for Ukrainian, but the guide explains all.

However, can you please send e-mail to the current translator (which I am still waiting a translation update from), to let him know that you are planning to work on it? You can find his e-mail in the About dialog when running Rufus in Ukrainian or by looking [here](https://github.com/pbatard/rufus/blob/6513b5e0b4b81c7d4c13ad9595bc5124ef92d57a/res/localization/rufus.loc#L16863).

Thank you. If you need help please e-mail me (I got your earlier e-mail), as I don't think the issue tracker is the best place to handle translation issues.

pbatard, 2018-05-14:
btw, latest version is __3.0__, NOT 3.1. Work for Rufus 3.1 has not started yet, and cannot start before 3.0 is released...

Alexander1414x, 2018-05-14:
Ok. I’m go test

Alexander1414x, 2018-05-14:
Please give me Rassian language , I’m go to edit Rassian to Ukraine.

pbatard, 2018-05-14:
I'm sorry but if you can't follow or understand the instructions from https://github.com/pbatard/rufus/wiki/Localization#Editing_an_existing_translation I'm going to have to decline your request to help with updating the Ukranian translation.

There's a single centralized `rufus.loc` file that contains all the translation. This is where you will find the Ukranian translation you need to modify and this is what you use to test. __All__ the instructions on how to accomplish that are in the Localisation guide I linked above. It's just not possible for me to hold your hand if you don't understand most of what you need to do from the guide, as I just don't have the time for that. Sorry.

Alexander1414x, 2018-05-14:
Im waiting.

Alexander1414x, 2018-05-14:
Ouch.
Im searched to problem localizate.
Please wait.

Alexander1414x, 2018-05-14:
Please Update : 
l "uk-UA" "Ukrainian (Українська)" 0x0422
v 3.0.22
b "en-US"

# Main dialog
g IDD_DIALOG
t IDS_DRIVE_PROPERTIES_TXT "Властивості приводу"
t IDS_DEVICE_TXT "Пристрій"
t IDS_BOOT_SELECTION_TXT "Тип завантаження"
t IDC_SELECT "Виберіть"
t IDS_IMAGE_OPTION_TXT "Опція зображення"
t IDS_PARTITION_TYPE_TXT "Схема розділу та тип системного інтерфейсу""
t IDS_TARGET_SYSTEM_TXT "Цільова система"
t IDC_LIST_USB_HDD "Показувати зовнішні USB накопичувачі"
# Note: It is acceptable to drop the parenthesis () below if you are running out of space
# as there is a tooltip (MSG_169) providing these details.
t IDC_OLD_BIOS_FIXES "Add fixes for old BIOSes (extra partition, align, etc.)"
# 'MBR': See http://en.wikipedia.org/wiki/Master_boot_record
# Rufus can install it's own custom MBR (the Rufus MBR), which also allows users
# to specify a custom disk ID for the BIOS. Note: the tooltip for this control is MSG_167.
t IDC_RUFUS_MBR "Використовуйте Rufus MBR з ідентифікатором BIOS"
t IDS_FORMAT_OPTIONS_TXT "Параметри форматування"
t IDS_FILE_SYSTEM_TXT "Файлова система"
t IDS_CLUSTER_SIZE_TXT "Розмір кластера"
t IDS_LABEL_TXT "Показник обсягу"
t IDC_QUICK_FORMAT "Швидкий формат"
t IDC_BAD_BLOCKS "Перевірте пристрій на погані блоки"
t IDC_EXTENDED_LABEL "Створюйте розширені мітки та піктограми"
t IDS_STATUS_TXT "Статус"
t IDCANCEL "Закрити"
t IDC_START "Старт"
`

Alexander1414x, 2018-05-14:
Im wait, update.

pbatard, 2018-05-15:
Sorry, but I don't think this can work. You don't understand what you need to do or how to test, and I don't have the time to explain that to you in details.

I'm going to have to decline your request. I do appreciate your request to help, but I also expect translators to be able to follow the localization guide. If that is not the case, it's simply going to be a waste of time for both of us, so I'd rather you don't work on the translation.

I'm going to close this request now.

closed 2018-05-15 by pbatard

Alexander1414x, 2018-05-15:
No

Alexander1414x, 2018-05-15:
вт, 15 мая 2018 г. в 10:40, Александр Кудрик <cawagood@gmail.com>:

> No
>
I’m working translate it is not problem .

Alexander1414x, 2018-05-15:
вт, 15 мая 2018 г. в 10:40, Александр Кудрик <cawagood@gmail.com>:

>
> вт, 15 мая 2018 г. в 10:40, Александр Кудрик <cawagood@gmail.com>:
>
>> No
>>
> I’m working translate it is not problem .
>
Please update . It is fix and normally .

pbatard, 2018-05-15:
Then provide me with a full `rufus.loc` file that is completely updated for Ukraine and that you have tested. Unless you can do that on your own, I really can't accept your work. Other translators have no issue doing just that, so I expect the same from you. If you can't do that on your own, then sorry, but I just won't be able to work with you.

Alexander1414x, 2018-05-15:
Okay Im test go. Please wait.

pbatard, 2018-05-15:
You don't have to reply. All I need is a completely updated `rufus.loc`. You should either send it through e-mail or create a pull request. I will not take bits and pieces of translations, and I do expect you to have looked at the [Changelog for 1.0.22 &rarr; 1.0.23](https://github.com/pbatard/rufus/blob/master/res/localization/ChangeLog.txt#L18-L68) to understand what you need to do.

Please don't reply until you can e-mail me a `rufus.loc` that has been fully updated for Ukrainian.

Alexander1414x, 2018-05-15:
Okay

Alexander1414x, 2018-05-15:
Please take 
[rufus.loc.txt](https://github.com/pbatard/rufus/files/2003931/rufus.loc.txt)

Please take  Im updated and fix this location.

Alexander1414x, 2018-05-15:
That's is right?

pbatard, 2018-05-15:
Please follow my instructions or I am just going to ignore you. I asked you to either send me your completely updated rufus.loc through e-mail or create a pull request. If you can't follow my instructions (and can't stop asking for update every other minute), I simply won't be able to work with you.

Alexander1414x, 2018-05-15:
Okay
wait. Im send.

pbatard, 2018-05-15:
You don't have to reply here. Just do what I ask!

pbatard, 2018-05-15:
1. I asked you to either e-mail me the file or create a pull request. You didn't do that
2. Your link produces a file that is not UTF-8. As a result it is __UNUSABLE__
3. Sorry but I cannot work with people who can't follow simple instructions.

Alexander1414x, 2018-05-15:
Ok. How to UTF-8 Code safe? Please gime me.

pbatard, 2018-05-15:
This will be my last reply in this thread unless you start to follow the instructions I give. As I told you 3 times already, I want you to e-mail me the file.

Alexander1414x, 2018-05-15:
Ok. Im send to mail.  No github.  Are you ok?

pbatard, 2018-05-15:
Gah, you are seriously way too needy!

As I answered to your e-mail, the file you sent does not work and it is also clear that you didn't test it as you should have, and that you don't have the ability required to complete this translation effort. I appreciate the willingness to help, but I'm afraid this simply can't work.

Please stop working on the Ukrainian translation, as I won't accept or answer any further request from you. Sorry.

Alexander1414x, 2018-05-15:
Ok. Im send file(this is UTF+8 without BOM) this is DOS/Windows mode . For edit Notepad 6.3, because this problem edit in 7.3 or lates't veriosn.
Im, sorry Im finish.  Are you is ok?