pbatard/rufus #1135: [Could not reset partitions

Qbsoon, 2018-05-15:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I'm trying to make bootable pendrive with rufus, but I get error. Sometimes error is "Cannot write", sometimes "Can't do partitioning". My first attemts were on 2.18 version of rufus, but now I updated to 3.0 BETA. When I first clicked start rufus downloaded newer version of Syslinux. Iso I'm using is "ubuntu-18.04-desktop-amd64" downloaded from ubuntu site.

Log
---
```
<FULL LOG>
Partition type: MBR, NB Partitions: 1
Disk ID: 0x2FFBF937
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.5 GB (15609102336 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 18.04 LTS amd64'
  Size: 1.8 GB (Projected)
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-18.04-desktop-amd64.iso (1.8 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20171017/' for Syslinux installation

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error [0x00000005] Odmowa dostępu.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not reset partitions
Re-mounted volume as 'G:' after error

Found USB 2.0 device 'USB Flash Memory USB Device' (0930:6545)
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1897, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x2FFBF937
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.5 GB (15609102336 bytes)
  Start Sector: 2048, Boot: Yes
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20171017/' for Syslinux installation

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error [0x00000005] Odmowa dostępu.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not reset partitions
Re-mounted volume as 'G:' after error

Found USB 2.0 device 'USB Flash Memory USB Device' (0930:6545)
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1897, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x2FFBF937
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.5 GB (15609102336 bytes)
  Start Sector: 2048, Boot: Yes

Computing checksum for 'D:\Obrazy\ubuntu-18.04-desktop-amd64.iso'...
  MD5:	 129292a182136a35e1f89c586dbac2e2
  SHA1:	 f373c0aec6162cdba76ee9084e695866a15e441a
  SHA256: a55353d837cbf7bc006cf49eeff05ae5044e757498e30643a9199b9a25bc9a34

pbatard, 2018-05-15:
I'm not seeing the bad blocks check from your log. Have you really run the bad blocks check? What did it report?

pbatard, 2018-05-15:
A few other things:

* According to your log, the error you get is `Error [0x00000005] Access is denied`. There's a dedicated [entry for that in the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied). Did you read it?
* Have you tried <kbd>Alt</kbd>-<kbd>,</kbd> as suggested by the FAQ?
* Also, what happens when you try to zero the drive (using <kbd>Alt</kbd>Z<kbd> &mdash; You can cancel it after a while if you don't see an error as this may take a long time)? Do you see the same error?
* You may also want to try to create a FreeDOS drive and see what happens there too.

Qbsoon, 2018-05-15:
Actually, I didn't saw that error. I was searching for solution to "could not reset partitions".

pbatard, 2018-05-22:
Since you haven't followed up on the advice I gave for one week, I'm going to have to assume that your problem was solved and close this issue.

closed 2018-05-22 by pbatard