pbatard/rufus #1142: create usb bootable from linux

candrapersada, 2018-06-02:
how to create usb bootable windows from linux?
and add create bootable without use usb or cd.

pbatard, 2018-06-02:
As clearly stated on the homepage, Rufus is a Windows application __only__.

If you want to create a bootable USB using Rufus, you must use a Windows machine.
And in case you want to ask, I am not planning to create a version of Rufus that runs on Linux:
https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxOS_XSome_other_OS

closed 2018-06-02 by pbatard

JonnyTech, 2018-06-02:
> how to create usb bootable windows from linux?

@candrapersada try [WoeUSB](https://github.com/slacka/WoeUSB)
(sorry @pbatard but leaving an answer in case others have the same question)

> and add create bootable without use usb.

Cannot understand what you are asking, but as Pete said: this is not the place for linux questions.

candrapersada, 2018-06-03:
add create bootable without use usb. use internal memory laptop