pbatard/rufus #1144: Suggestion: Make group policy modifications opt-in

r2987554, 2018-06-03:
Since keeping autorun turned off seems to be fairly common and recommended as part of improving Windows security since years, I think that disabling it via Group Policy on every launch of Rufus is a bit of an overkill and could be made opt-in like the automatic update. That would also eliminate risk (and mostly, fear) of any related issues.

pbatard, 2018-06-03:
Not gonna do that. A lot of Rufus users will not understand what this is about, and it will be a major inconvenience for most. Plus there's nothing sneaky about disabling autorun as it actually makes one's computer safer. &rarr; Closed.

But if you can think about actual "related issues", rather than mere fear of it, I'll be happy to hear you.

closed 2018-06-03 by pbatard

r2987554, 2018-06-03:
_Plus there's nothing sneaky about disabling autorun as it actually makes one's computer safer._

That's exactly what I meant: if a person knowingly disabled autorun years ago, there's no need for any app to disable it again. Options for Rufus could be simple:
**Temporarily turn off autorun popups (Recommended)** "Leave this unchecked if autorun for removable media is disabled in your system"