pbatard/rufus #1149: Latest version 3.0.1304 issue with installing Windows 10 (1709)

NVader2000, 2018-06-08:
We have been using version 2.18 for a long time without issue when burning Windows 10 build 1709. We have used both the ISO from or VLA and the Windows Media Creation Tool and both work fine. We have used both MBR and GPT on various device and all have worked fine.

However, with the latest version of Rufus 3.0.1304 there are issues where either Windows reboots after completing and boots off the USB and starts the install process all over again.  Other issue includes SecureBoot issues on Surface Pro 4.

Returning to version 2.18 resolves all of these issues.

pbatard, 2018-06-08:
That is very surprising considering that the internal code to created Windows bootable USBs has not changed at all. Are you sure you aren't being tripped by the fact that the default for Windows images in 2.18 is to use BIOS or UEFI-CSM settings and the default for 3.0 is to use pure UEFI? Obviously, if you have a target with CSM enabled, the differences you may be seeing could boil down to a UEFI specific behaviour with your platforms.

Still, to investigate this issue, I will need you to attach both the __FULL__ logs of the (working) USB you create with 2.18, and the (non-working) USB you create with 3.0. I absolutely must be able to compare your logs to investigate this issue.

pbatard, 2018-06-08:
Just for the record, we also performed a __recursive file comparison__ of 2 USB flash drives created with `en_windows_10_multi-edition_version_1709_updated_sept_2017_x64_dvd_100090817.iso` (jn GPT/UEFI mode), one with Rufus 2.18 and one with Rufus 3.0, and found no difference in the binary content of the files whatsoever and no file missing from one drive or the other.

Considering that UEFI does not use anything like bootloaders, and simply looks at the file contents of a partition with a supported file system (such as FAT32 as we used here), I hope you will agree that this means that, if you are seeing differences of behaviour in UEFI mode, then they have nothing to do with the version of Rufus you are using.

Also, please be mindful that, in hundreds of thousands of downloads of Rufus 3.0, a lot of which are being used to install Windows, you are the only person to report a behaviour that, by all means, should be resulting in a very large number of issue reports, which we aren't seeing at all.

All in all, I have to conclude that the issue you report has to do with something on your side, and not with Rufus.

pbatard, 2018-06-11:
No further update from OP &rarr; Closed

closed 2018-06-11 by pbatard