pbatard/rufus #1150: Rufus only formats my USB

tiagossa1, 2018-06-08:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

It is my first time here, so I'm sorry if I'm doing something wrong.
I've updated to newest Rufus 3.0. I was making a bootable USB of Manjaro XFCE, so I download it , plugged USB, chose Manjaro's ISO and then press "Start". It only formats my USB, nothing else.
There's the log attached.
Rufus 2.8 works fine, so I'm assuming it's something with this new version.
[rufus.log](https://github.com/pbatard/rufus/files/2084196/rufus.log)

pbatard, 2018-06-08:
Duplicate of #1145. This is a known issue with Grub based ISOs like Manjaro. It will be fixed in the next version. In the meantime, please write the image in DD mode when Rufus asks you (remember, Rufus explicitly tells you that if your USB doesn't work in ISO mode, you should retry writing in in DD mode), or use 2.18 is you really want to write it in ISO mode.

closed 2018-06-08 by pbatard

tiagossa1, 2018-06-08:
Thanks! So sorry, i didn't saw that issue before! :)

pbatard, 2018-06-08:
Yeah. I reckon it is a bit difficult to track down, which is why I changed the title from yours to explicitly mention Manjaro, so it will help others running into the same problem.

And just in case you wonder, Rufus 3.1, with the fix for that issue, should be released sometime before the end of the month.