pbatard/rufus #1152: RUFUS HAS DAMAGED A SD CARD

gabrielcvas, 2018-06-09:
Hello everyone, 

I've borrowed a SanDisk SD Card (32gb of capacity) from a friend of mine to create a bootable windows installation drive. I ran a test on the sd card and it was fine (I formated the drive via windows explorer and it was showing 29,5gb of disk capacity, which is normal). 

I've used rufus many times and I never had any issue until now. 

The installation failed and the SD Card is now showing **2 terabytes of capacity** (I don't know how it is even possible to do that). So far i've done this to undo rufus' actions on the SD Card: 

- Tried to format through windows explorer (failed, still showing 2TB)
- Tried to format with EASEUS (failed, still showing 2TB)
- Tried to run chkdsk /f (failed, it said it cannot run under RAW partitions)
- Tried to use windows disk manager and it showed 2tb of capacity as well. 

I ran out of options... I need help...

pbatard, 2018-06-09:
Most likely the controller in your SD card detected that the flash memory was defective (flash memory has a limited number of write cycles, and, depending entirely on your luck, may happen to fail at any random moment) and has reverted to a set of default settings from the controller. This is probably why, since it can no longer access the flash at all, it is reporting the maximum capacity it was programmed for. Of course, only the people who designed the microcontroller that accesses the flash card can tell you what the real story is.

However, when flash memory fails (which does happen regardless of the application being used to access it), you will typically get into the situation where the media behaves in strange fashion, such as reporting that there is no card inserted (even if there is no card to removed) or that it is read-only or reporting an erroneous capacity.

At this stage, the best thing to do is purchase a new card, and remember that, even more than HDDs, flash memory will ultimately fail.

pbatard, 2018-06-11:
I'm going to consider that the answer above addressed your query and close this issue.

closed 2018-06-11 by pbatard