pbatard/rufus #1153: [Suggestion] Update speed comparison

Pulsera, 2018-06-11:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
---------

The speed comparison in your website and the README file are completely outdated. It would be a good idea considering Rufus had major changes since v1.1.0, that was 6 years ago.
Also the UNetbootin link redirects to the new official link, which is [https://unetbootin.github.io/](https://unetbootin.github.io/) and the versioning for their program appears to be different. Latest one is 661. According to properties the software is still v1.1.1.1 but for some reason they never changed that.
Also the windows link, loads the microsoft website, which tells you to download from codeplex, which redirects you to archive.codeplex which tells you the project is on a microsoft link. It would be better to just link the file download than going through several links.

- Test new Rufus release (v3.0)
- Update UNetbootin link with the newer one. https://unetbootin.github.io
- Test newer release of UNetbootin (661)
- Update Windows link. https://www.microsoft.com/en-us/download/details.aspx?id=56485
- Test with Windows 10, Ubuntu 18.04, and any other OS that seems relevant (Maybe BSD one)

pbatard, 2018-06-11:
Yeah, I've been having some plans to update the speed comparison, but it's just not a high priority.

Also, even though the application changed here and there, internally, the ISO &rarr; USB extraction has hardly been altered in a way that should affect the speed.

I'm going to keep that issue, but flag it as deferred.

pbatard, 2018-06-11:
PS: I have updated the `Windows 7 USB download tool` link on the homepage to point to the official Microsoft information page (which may redirect to codeplex archive for the download link, but that's entirely Microsoft's issue, as the goal from my link is to provide a description of what the tool does, not a download link).

shwegyi198361, 2020-07-03:
> Yeah, I've been having some plans to update the speed comparison, but it's just not a high priority.
> 
> Also, even though the application changed here and there, internally, the ISO → USB extraction has hardly been altered in a way that should affect the speed.
> 
> I'm going to keep that issue, but flag it as deferred.

shwegyi198361, 2020-07-03:
> ## Checklist
> * [x]  I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
> * [x]  I performed a search in the issue tracker for similar issues, using [...]

oam7575, 2022-09-30:
Hello All.

I have carried out some benchmarking of Rufus 3.20.1929 vs newer versions of several (somewhat) comparable applications.
To make testing faster and remove as many bottlenecks as possible source drive was internal NVMe SSD target drive was USB 3.0 SSD.

Details and expanded notes are below.

Crystal Diskmark shows about 250Mbytes per second for the USB SSD

[image]

### Results Table

<html>
<body>

**Windows 7** | **Minutes** | **Seconds** | **Notes**
-- | -- | -- | --
Rufus 3.20.1929 | 0 | 41 |  
RMPrepUSB 2.1.746 | 0 | 44 |  
Universal USB Installer 2.0.1.4a | 0 | 32 |  
Unetbootin 702 | 1 | 30 |  
MS Media Creation Tool | NA | NA |  
Etcher 1.7.9 Portable | NA | NA |  
  |   |   |  
**Windows 10** | **Minutes** | **Seconds** |
Rufus 3.20.1929 | 0 | 43 |  
RMPrepUSB 2.1.746 | 0 | 44 |  
Universal USB Installer 2.0.1.4a | 0 | 32 |  
Unetbootin 702 | 1 | 30 |  
MS Media Creation Tool | 1 | 5 | Approximate time after ISO download and verify process completed
Etcher 1.7.9 Portable | NA | NA |  
  |   |   |  
**Windows 11** | **Minutes** | **Seconds** |
Rufus 3.20.1929 | 1 | 13 |  
RMPrepUSB 2.1.746 | 0 | 42 |  
Universal USB Installer 2.0.1.4a | 0 | 31 |  
Unetbootin 702 | 1 | 31 |  
MS Media Creation Tool | NA | NA |  
Etcher 1.7.9 Portable | NA | NA |  
  |   |   |  
**Ubuntu 22.04.1 LTS DVD** | **Minutes** | **Seconds** |
Rufus 3.20.1929 – ISO MODE | 0 | 35 |  
Rufus 3.20.1929 – DD MODE | 0 | 18 |  
RMPrepUSB 2.1.746 | 0 | 50 |  
Universal USB Installer 2.0.1.4a | 0 | 20 |  
Unetbootin 702 | 1 | 12 |  
MS Media Creation Tool | NA | NA |  
Etcher 1.7.9 Portable | 0 | 34 |  
  |   |   |  
**Slackware 15 x64 DVD** | **Minutes** | **Seconds** |
Rufus 3.20.1929 – ISO MODE | 1 | 19 |  
Rufus 3.20.1929 – DD MODE | 0 | 18 |  
RMPrepUSB 2.1.746 | 1 | 27 |  
Universal USB Installer 2.0.1.4a | 0 | 25 |  
Unetbootin 702 | 8 | 30 |  
MS Media Creation Tool | NA | NA |  
Etcher 1.7.9 Portable | 0 | 36 |  

</body>

</html>

### Hardware and Software
<html>
<body>

**Motherboard** : Gigabyte X570 UD
**CPU** : Ryzen 5 3600
**RAM** : 16GB
**OS** : Win 10 Pro 21H2 x64
**Source Drive** : Sasmsung 970 EVO Plus 500Gb NVMe
 
**Destination Port** : USB 3.0 SuperSpeed
**USB Housing** : AS2105 USB 3.0 SATA
**Destination Drive** : Samsung 840 SSD SATA

 
**Windows 7 SP1 Ultimate ISO**
https://download.microsoft.com/download/5/1/9/5195A765-3A41-4A72-87D8-200D897CBE21/7601.24214.180801-1700.win7sp1_ldr_escrow_CLIENT_ULTIMATE_x64FRE_en-us.iso
 
**Windows 10 21H2**
https://software.download.prss.microsoft.com/dbazure/Win10_21H2_English_x64.iso?t=b94f1692-c9ef-47bd-97d2-6404067bd926&e=1663845512&h=bc32037ea38c54efca55ff53988275e02f673e99681fa040325bb5866fc4667a
 
**Windows 11**
https://software.download.prss.microsoft.com/dbazure/Win11_English_x64.iso?t=9310918c-5355-4e48-9120-8ca26d7d3cc4&e=1663845859&h=302c3f68b1f64a8ca7d535242746ca8a4913e8144123132be64e3b9ac7b6d951
 
**Ubuntu 22.04.1 Desktop**
https://releases.ubuntu.com/22.04/ubuntu-22.04.1-desktop-amd64.iso
 
**Slackware 15 x64**
https://mirrors.slackware.com/slackware/slackware-iso/slackware64-15.0-iso/slackware64-15.0-install-dvd.iso
  
**Tools Used**
**Windows 10 21H2 Media Creation Tool**
https://download.microsoft.com/download/b/0/5/b053c6bc-fc07-4785-a66a-63c5aeb715a9/MediaCreationTool21H2.exe
 
**Universial USB Installer : 2.0.1.4a.exe**
https://www.pendrivelinux.com/downloads/Universal-USB-Installer/Universal-USB-Installer-2.0.1.4a.exe
 
**Unetbootin : https://unetbootin.github.io/**
https://github.com/unetbootin/unetbootin/releases/download/702/unetbootin-windows-702.exe
 
**RMPREPUSB**
https://rmprepusb.com/downloads/RMPrepUSB_Full_v2.1.746.exe
 
**Blana Etcher**
https://www.balena.io/etcher/
https://github.com/balena-io/etcher/releases/download/v1.7.9/balenaEtcher-Portable-1.7.9.exe

</body>

</html>

Final note, I am aware of Ventoy which essentially uses a copy / paste of a complete ISO to create a multi boot drive.
I used a batch script to time a dos / cmd level file copy operation for some of the ISO's listed above.
The batch script was taken from here : https://stackoverflow.com/questions/673523/how-do-i-measure-execution-time-of-a-command-on-the-windows-command-line#

**Results**

timecmd.bat copy Win11_English_x64.iso i:\
        1 file(s) copied.
command took 0:0:24.34 (24.34s total)

timecmd.bat copy slackware64-15.0-install-dvd.iso i:\
        1 file(s) copied.
command took 0:0:16.69 (16.69s total)

timecmd.bat copy ubuntu-22.04.1-desktop-amd64.iso i:\
        1 file(s) copied.
command took 0:0:16.83 (16.83s total)

pbatard, 2022-10-01:
Thanks for this extensive report.

I'm happy to see that Rufus is killing it for DD mode, which demonstrates that the changes we applied in a0904cad354fb1d97fb1313b598beeb4b038fc6c were worth it.

And those changes is pretty much the reason why we haven't published updated results, because we are planning to apply similar speed improvements for ISO mode, which we know have been lacking a bit. Oh, and for the record (at least this is something I've seen for the MD5/SHA checksum computations) I do believe that Rufus is probably also penalized by the fact that, unless you're running the Windows Store version, you're running a 32-bit application and that will add some overhead for I/O, whereas utilities like UUI use 64-bit version of 7-zip for extraction, which does help (though 7-zip on its own, is already pretty much the best in class for data extraction).

All this too say that we're well aware that there is room for improvement, which we're planning to accomplish by switching ISO I/O to async mode (item 2 on #1646), just like we did switch DD I/O to async mode with the good results the benchmarks above have shown. However, this will take some time and some planning, which is evidenced by the fact that we've been planning on carrying this out for a few years now, and still haven't managed to allocate the time to do so.

And that is the precise reason why, until we have integrated these planned I/O changes, we're not going to update the speed comparison reports, because we don't consider that these results will be representative of where Rufus will be once our I/O has been finalized.

Finally, just a small nitpick: I don't think splitting minutes and seconds in a table is for the best, as it makes it harder to get the full picture at a glance. I think those times would be better to reside in a single column alone, either as a `mm:ss` value or just a `sss`.
But once again, thanks a lot for spending time on this!

pbatard, 2023-04-22:
I'm going to close this issue, on account that I have now removed the speed comparison from the homepage due to the fact that it's become mostly pointless when most people will be comparing Rufus with Ventoy, and we're never going to be "best" Ventoy since we're extracting ISO content rather than copying it wholesale. In other words, because any comprehensive speed comparison table now implies comparing apples with oranges, there's just no point in trying to provide one.

I am still very much planning to apply the double buffering changes to ISO file extraction when I get the chance, and of course apply any changes that can improve Rufus write speeds, but I think the speed comparison has run its course...

closed 2023-04-22 by pbatard