pbatard/rufus #1155: Device in use by another process!

sanchit-beginnings, 2018-06-11:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [x] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I get this error and the log is reported below. This is brand new USB I bought yesterday. Also, I have tried disabling the McAfee, but still it doesn't work. Also, I have separately checked MD5SUM check for my ISO file. Please help!

Log
---
```
Rufus version: 3.0.1304
Windows version: Windows 10 64-bit (Build 16299)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1905, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0614270F
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15668871168 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20151222 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 16.04.4 LTS amd64'
  Size: 1.5 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-16.04.4-desktop-amd64.iso (1.5 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20151222/' for Syslinux installation

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
No drive letter was assigned...
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error [0x00000002] The system cannot find the file specified.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not reset partitions
Re-mounted volume as 'D:' after error

0 devices found
```

pbatard, 2018-06-11:
Your issue title does not match the error from the log. Also, what was the report of the bad blocks check? Can I get the log with the bad blocks check? There are a few inconsistencies in your report that make it very difficult to troubleshoot what your issue might be. Especially is it exceedingly surprising that the bad blocks check would have worked if you get a write error when clearing the MBR/GPT records...

If you really tested your flash drive on a different computer and got the same issue, then I'm afraid it's either a hardware problem, or you have some application installed on both computer, that you will need to find, that is prohibiting low level access. You may want to check the list of software that will interfere with Rufus [at the end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software).

pbatard, 2018-06-13:
No update from OP &rarr; Closed.

closed 2018-06-13 by pbatard

buttererror, 2018-10-03:
I had the same problem but i didn't read the log, however, you can try clearing your USB label and try again, or format your USB as normal with windows, then use the rufus with out "quick format" checked.

joeyc002, 2019-04-04:
Had same issue with Rufus on 3 machines each with a different pen drive. ISO to USB worked fine however.