pbatard/rufus #1156: Windows cannot read my USB after trying to install Porteus ISO using Rufus

Skazee, 2018-06-12:
Im using Rufus 3.0.1304 Portable

The USB Flash disk is SanDisk Cruzer Blade 128GB which a new one(fresh out from its packaging)

Before proceed with a ISO burn process Rufus, I had try to check the USB Flash disk via My Computer... and its working as Windows 7(Yups Im using Win 7) can read it.

While on process of burning the Porteus ISO... Rufus encountered some problem(which Im already forgetten what it is)... and thus it doesnt completed the process..

Then I realized that my USB aint working anymore... Try checking in diskmgmt.msc and its detecting theres no media on that drive... Try changing the drive letter from (H to Z) and its doesnt solved my issue.
Try connecting it to other laptop(2 laptop in this case) the same problem arise.

Skazee, 2018-06-12:
Thank pbatard... 

Im also suspecting that this USB Flash Disk somehow is a fake one... but none the less... could you point me where to find the rufus log?

pbatard, 2018-06-12:
Yeah, fake drives are unfortunately very real, and of course, the unscrupulous people who produce them will target reputable brands and large capacity drives, so your SanDisk Cruzer Blade 128GB would be a prime target for that.

If the price you got for your drive seemed like a great bargain, chances are you got a fake drive.

With regards to your question, the log is not saved between session (because Rufus tries to be nice and not leave potentially unwanted files behind), so you won't be able to access the log from the time of failure. Still. you can access the log while Rufus is running by clicking the last small button, left of the _START_ button.

pbatard, 2018-06-14:
Since I don't think there's anything more I can do, I will close this issue.

closed 2018-06-14 by pbatard