pbatard/rufus #1157: Kaspersky Rescue 2018 with Rufus latest.

Vasilx14, 2018-06-12:
Problem Kaspersky Rescue 2018 write in ISO recommend.

Found USB 2.0 device 'SanDisk Ultra USB 3.0 USB Device' (0781:5591)
1 device found
Disk type: Removable, Disk size: 16 Гб, Sector size: 512 bytes
Cylinders: 1869, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x2044524B
Drive has an unknown Master Boot Record
Partition 1:
  Type: EFI FAT (0xef)
  Size: 2.8 MB (2949120 bytes)
  Start Sector: 1124280, Boot: No

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE2 for exclusive write access
Requesting lock...
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Grub 2.0 MBR
Writing Grub 2.0 SBR (from embedded)
Found volume GUID \\?\Volume{04ea85b3-56af-11e8-b453-240a6474715a}\
Waiting for access on \\?\Volume{04ea85b3-56af-11e8-b453-240a6474715a} [\Device\HarddiskVolume8]...
Warning: Could not obtain exclusive rights. Retrying with write sharing enabled...
WARNING: The following process(es) or service(s) are accessing \Device\HarddiskVolume8:
● 'C:\Windows\explorer.exe' (pid: 1536, access: r)
You should close these applications before attempting to reformat the drive.
Opened \\?\Volume{04ea85b3-56af-11e8-b453-240a6474715a} for shared write access
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Successfully remounted \\?\Volume{04ea85b3-56af-11e8-b453-240a6474715a}\ on F:
Finalizing, please wait...
Created: F:\autorun.inf
Created: F:\autorun.ico
Successfully remounted \\?\Volume{04ea85b3-56af-11e8-b453-240a6474715a}\ on F:

Found USB 2.0 device 'SanDisk Ultra USB 3.0 USB Device' (0781:5591)
Using autorun.inf label for drive F: 'KRD'
1 device found
Disk type: Removable, Disk size: 16 Гб, Sector size: 512 bytes
Cylinders: 1869, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0011C477
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.3 GB (15375269888 bytes)
  Start Sector: 2048, Boot: Yes

[image]

Vasilx14, 2018-06-12:
And I want to go This DD  this is not problem.

Vasilx14, 2018-06-12:
Im return in version 2.18  this is not problem ,because this is bug 3.0 .

Im testet , this is not problem write 2.18 .  Please see log:  
Rufus version: 2.18.1213
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0419
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
NOTE: The Russian (Русский) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'SanDisk Ultra USB 3.0 USB Device' (0781:5591)
1 device found
Disk type: Removable, Disk size: 16 Гб, Sector size: 512 bytes
Cylinders: 1869, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x2044524B
Drive has an unknown Master Boot Record
Partition 1:
  Type: EFI FAT (0xef)
  Size: 2.8 MB (2949120 bytes)
  Start Sector: 1124280, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected Grub version: 2.02
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'KRD'
  Size: 548.3  Мб (Projected)
  Uses: GRUB2
  Uses: EFI
Using image: krd.iso (551.8  Мб)

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE2 for exclusive write access
Requesting lock...
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Grub 2.0 MBR
Writing Grub 2.0 SBR (from embedded)
Found volume GUID \\?\Volume{04ea85b3-56af-11e8-b453-240a6474715a}\
Waiting for access on \\?\Volume{04ea85b3-56af-11e8-b453-240a6474715a} [\Device\HarddiskVolume8]...
Opened \\?\Volume{04ea85b3-56af-11e8-b453-240a6474715a} for exclusive write access
Writing partition boot record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
[... 946 lines cut ...]
Extracting: F:\data\005-bases.srm (124.8 MB)
Extracting: F:\data\005-bases.srm.sha512 (144 bytes)
Extracting: F:\data\008-firefox.srm (50.6 MB)
Extracting: F:\data\008-firefox.srm.sha512 (146 bytes)
Extracting: F:\data\kernel.dat (125.9 MB)
Extracting: F:\data\kernel.dat.sha512 (141 bytes)
Extracting: F:\efi.img (2.8 MB)
Extracting: F:\krd_bases_timestamp.txt (15 bytes)
Extracting: F:\krd_version.txt (66 bytes)
Extracting: F:\mach_kernel (0 bytes)
Finalizing, please wait...
F:autorun.inf already exists - keeping it

Found USB 2.0 device 'SanDisk Ultra USB 3.0 USB Device' (0781:5591)
Using autorun.inf label for drive F: 'KRD'
1 device found
Disk type: Removable, Disk size: 16 Гб, Sector size: 512 bytes
Cylinders: 1869, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001ADEBA
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.3 GB (15375269888 bytes)
  Start Sector: 2048, Boot: Yes

pbatard, 2018-06-12:
Duplicate of #1145. This has already been fixed in the source and the fix will be available in next version.

closed 2018-06-12 by pbatard