pbatard/rufus #1159: cannot format usb

chaitanyajasthi, 2018-06-16:
Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2018-06-16:
Did not follow the check list &rarr; Closed.

What was the result of the bad blocks check?

closed 2018-06-16 by pbatard

chaitanyajasthi, 2018-06-18:
> Did not follow the check list → Closed.
>
> What was the result of the bad blocks check?

no bad blocks

pbatard, 2018-06-18:
`I/O device error` means that you have a hardware issue. This cannot be fixed in the software, so I'm afraid you will have to find out why that is the case on your own. For instance, I'm pretty sure you will get the same error if you try to use Windows `diskpart` to clean and format your drive or Windows explorer to copy files onto your USB. It just not possible for Rufus to produce an `I/O device error` unless other applications have the same issue.

chaitanyajasthi, 2018-06-18:
Yes, I get the same error when formatting from disk partition
And I don't get this error with other usb

pbatard, 2018-06-18:
Which means your USB Flash Drive is defective. There's really nothing Rufus can do for you there.