pbatard/rufus #1160: Error While partitioning drive

ChocolateBoyChiku, 2018-06-16:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
i tried it many times it worked fine before but now it promted with this report

Log
---
Rufus version: 3.0.1304
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'MXT-USB Storage Device USB Device' (AAAA:8816)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 480, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT32 (0x00)
  Size: 3.7 GB (3955228672 bytes)
  Start Sector: 0, Boot: No
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 480, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT32 (0x00)
  Size: 3.7 GB (3955228672 bytes)
  Start Sector: 0, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'CDROM'
  Size: 49.9 MB (Projected)
  Note: File on disk is larger than reported ISO size by 6.7 MB...
  Uses: Syslinux/Isolinux v6.03
Using image: mini.iso (57 MB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20171017/' for Syslinux installation

Format operation started
Requesting disk access...

pbatard, 2018-06-16:
Does your log __really__ end with `Requesting disk access...`. I find this very surprising, because then that would mean that Rufus freezes, even though there is a timeout in the code for that operation..

If not, please make sure to attach the full log, with the part that is displayed __after__ `Requesting disk access...`, which should display an error. Without seeing that error, I can't help you.

ChocolateBoyChiku, 2018-06-16:
[image]

ChocolateBoyChiku, 2018-06-16:
Rufus version: 3.0.1304
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'MXT-USB Storage Device USB Device' (AAAA:8816)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 480, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT32 (0x00)
  Size: 3.7 GB (3955228672 bytes)
  Start Sector: 0, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'CDROM'
  Size: 49.9 MB (Projected)
  Note: File on disk is larger than reported ISO size by 6.7 MB...
  Uses: Syslinux/Isolinux v6.03
Using image: mini.iso (57 MB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20171017/' for Syslinux installation

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE2 for exclusive write access
Requesting lock...
Will use 'H:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not reset partitions
Re-mounted volume as 'H:' after error

Found USB 2.0 device 'MXT-USB Storage Device USB Device' (AAAA:8816)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 480, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT32 (0x00)
  Size: 3.7 GB (3955228672 bytes)
  Start Sector: 0, Boot: No

pbatard, 2018-06-16:
There's an entry for `[0x00000005] Access is denied` [in the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied).

You have software installed that is actively preventing Rufus to perform its job. It's up to you to identify and disable it.

closed 2018-06-16 by pbatard