pbatard/rufus #1162: Revert to old GUI?

Jeansen, 2018-06-19:
First I would like to say that I am really happy this tool exists. Please keep up the good work! That said, I have one little issue regarding the new GUI. In my humble oppinion, the older GUI was mucht cleaner, top down; simple. The new GUI simply feels ..... unfinished, untidy, without order. I would like to see the 2.x GUI again. 

But, this only counts for me. Maybe if other users would vote on my poste here (thumbs up/down), a better assesment would be possible on what the community thinks about this topic ;)

Cheers,
Marcel

pbatard, 2018-06-19:
Not gonna happen. This is a purely subjective matter, where I am of course taking __opinions__ into account, but that is not going to be subject vote. 

I also have to state that, so far, feedback for the new UI has been overly positive (you are the only second dissenting voice I heard since the 3.0 release, and I've seen way many more people express the view that this was indeed a much better design than the old version).

Therefore, considering that, since the feedback I have received has not been overwhelmingly negative during the month where the new design has been made public (which is the only way I would have considered reverting the design), I am going to close the issue, especially as internet polls can too easily be manipulated and, at this stage, I can safely consider that I have gotten enough feedback with regards to the new GUI, to be confident that you are indeed in the minority if you want to see it reverted.

Then again, the beauty of Open Source is that, if you think that the new design sucks, and want to see a version of Rufus that sticks with the old one, there's nothing preventing you from creating a fork that does just that, and then distributing that version.

Of course, doing so may also help you get a better understanding of whether there are actually as many people as you believe who find the old UI better, coz, if that is the case, you should have no trouble at all finding a sympathetic group of developers who want to create a fork of Rufus based on the old UI. For instance,  when people reacted against the introduction of `systemd` in Debian, a Devuan fork without `systemd` was quickly established, and projects do fork all the time on disagreements, so what I'm talking about here is far from an exceptional occurrence.

I should also point out that, as much as many people seem to be believing so, an Open Source project is not a democracy and therefore I, as the project maintainer, will make no qualms attributing myself some freedoms that you may not agree with when it comes to __subjective__ elements (because these are elements you'll have a lot of trouble making a logical case against). However, if you disagree with my decisions, you are never ever stuck with having to follow my lead. Instead, you can always find a bunch a developers that agree with you, and create your version. Then, if you are "right" and I am "wrong", you should have no trouble rallying people to your cause and become __the__ main repository for the project. 

IMO, that's a lot better than trying to rely on internet votes (as well as forcing a developer to go through something they disagree with for a lot of reasons).

closed 2018-06-19 by pbatard

Jeansen, 2018-06-20:
@pbatard Thank's for your thorough statement. I appreciate that (and all the work you put into rufus). Your decision is fine with me. As I said: It is my truly personal opinion. It is not meant as a general criticism. I totally agree with your standpoint. And if - so far - there has been only a minority of people who wish the GUI would be different then it is also an indication to keep things as they are.

But that is exactly the beauty of open source. You, the project maintainer, get feedback that you can work with. Either you will change something (or allow the community to change something), or you don't. And if people still disagree, then they can fork a project and do it their way. Though I would like to see more efforts in having one project flourish, than a million forks because people disagree on minor details.

As a last word: I know there is a lot of gossip regarding systemd. And yes, it broke with one major philosophy of *NIX (as did GNOME and others) but there is also a lot good in it. And in the end, it is the community that decides. If they all want it, they will use it. If not, the will use something else.

Anyway, I just wanted to give you some feedback. Nothing more, nothing less :smile: 

So, please keep up the good work. Rufus is a great piece of software!