pbatard/rufus #1163: [Enhancement] Support for wiping disks?

rmenessec, 2018-06-19:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Would you consider adding support for wiping disks during format? Nothing fancy; just zeroing the whole device as fast as possible. I'd most like to be able to issue wipe-while-format for HDDs, but possibly SSDs as well.

Post-erase, sending TRIM to SSDs and DISCARD* to SD cards that support it would be fantastic.

For that matter, having a separate option to insecurely "erase" SSDs and SD cards by marking all space as free with TRIM / DISCARD would be great.

* For details on DISCARD, as well as FULE (Full User area Logical Erase), see "[SD Specifications Part 1: Physical Layer Simplified Specification](https://www.sdcard.org/downloads/pls/)" Version 6.00, 4.3.5.2 - 4.3.5.3 / pp 40-42.

Log
---
```
Rufus version: 3.1.1320
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found non-USB removable device 'SAMSUNG SSD 830 Series' => Eliminated
Found non-USB removable device 'Crucial_CT512MX100SSD1' => Eliminated
Found USB device 'Samsung Flash Drive USB Device' (090C:1000) [ID]
Using autorun.inf label for drive E: 'Samsung MUF-64BA-AM'
1 device found
Disk type: Removable, Disk size: 64GB, Sector size: 512 bytes
Cylinders: 7799, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00357F21
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Size: 59.7 GB (64155025408 bytes)
  Start Sector: 2048, Boot: No

```

pbatard, 2018-06-19:
I know the FAQ is quite long, but next time, please spend a bit more time reading through it.
<kbd>Alt</kbd>-<kbd>Z</kbd> is what you are looking for (see [here](https://github.com/pbatard/rufus/wiki/FAQ#Power_keysCheat_modes).

Or you can uncheck _Fast format_ in the _advanced format options_ (this will zero all the blocks from the file systems that aren't use).

Or you can run multiple passes of the bad blocks check, since it will also fill every single accessible block from your USB with data, erasing whatever was there before.

In other words, you already have more than one option to achieve what you want.

closed 2018-06-19 by pbatard

rmenessec, 2018-06-19:
I'm sorry if you feel like I've wasted your time, but I _did_ search the FAQ for "erase," "wipe," "overwrite," "unrecover," and "TRIM," and found nothing that clearly indicated when Rufus does or does not deliberately _make unrecoverable_ all data on a target volume. "Erase" as used in the FAQ could simply mean that an MBR or partition table is lost.

Does Rufus have support for TRIM or DISCARD?

pbatard, 2018-06-19:
> Does Rufus have support for TRIM or DISCARD?

No. It's pointless for the vast majority of USB flash drives that Rufus is meant to be used with, and there just isn't enough cases where that would be useful. If you need to enact TRIM, I will respectfully ask that you use a dedicated tool to do just that, as this is way beyond the scope I have set for Rufus.