pbatard/rufus #1164: After failed use, my USB doesn´t get recognized

GeekBro17, 2018-06-20:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<First, I hope to express well (not English native)
Ok, here´s the thing: I was burning an Iso Image to my usb, all normal, as always. Then,
accidentally the USB got unplugged and obviously the process got stopped and failed (I have to add that the USB did not harm physically). I plugged it again and the USB does not get recognized by the pc, it says like "There´s no disk on this divice" or stuff like, I try to format it and it says "This device is not formattable", it also says that there´s no media!
What is it probably? What could I do? >

Log
---
```
<Rufus version: 2.18.1213
Windows version: Windows 8.1 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
Localization set to 'en-US'
Found USB 2.0 device 'Generic USB Flash Disk USB Device' (058F:1234)
Device eliminated because it appears to contain no media
0 devices found>
```

pbatard, 2018-06-20:
Answered in the FAQ:
https://github.com/pbatard/rufus/wiki/FAQ#Rufus_cant_detectformat_my_drive_and_reports_that_it_is_readonly_or_contains_no_media

There's not much Rufus can do if your device is not recognized, and I'm afraid you would have had the __exact__ same issue if you had unplugged your drive while you were copying a large folder using Windows Explorer, because that's basically what Rufus was doing (and Rufus is not doing anything different from what Windows Explorer does while copying files).

Looks like the controller inside your USB brought the flash memory offline for whatever reason. This is not something Rufus was responsible for, or something that Rufus can help in recovering. Most likely, you will have to purchase a new drive (and make sure you don't unplug this new drive when you shouldn't!).

pbatard, 2018-06-25:
Since there has been no upate from OP, I will close this issue.

closed 2018-06-25 by pbatard