pbatard/rufus #1165: I am unable to run the .exe of 3.1. Every download I try, it seems like it is not a valid Windows file. Not sure how to fix, nor do I know if I am the only one experincing the issue, but I wanted to let you know

kaoscoach, 2018-06-20:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2018-06-20:
If you are running Windows Xp or Windows Vista, please be mindful that, as mentioned multiple times on the [homepage](https://rufus.ie), Rufus 3.x is __NO LONGER__ compatible with these platforms, and what you describe is exactly what's expected to happen there.

Oh, and in case you wonder, I looked into a way to add a message that would say something like _This version of Rufus is not compatible with this platform_ when launching it on older version of Windows, but the problem is that, the whole point of dropping XP and Vista is so that I could upgrade the default libraries being used on Rufus launch, so that I can use more modern APIs, which of course prevents the application from launching altogether on old platforms... and therefore display a message that says that Rufus is incompatible. And I just don't have the time to add a massive kludge to Rufus so that it remains compatible with old versions of Windows just so that it can display a message stating that it is not compatible.

So, what version of Windows were you trying to run Rufus on?...

kaoscoach, 2018-06-20:
I am using Windows 10. I have tried it on two separate systems in two different locations.

pbatard, 2018-06-20:
Strange. Nobody else but you is reporting it so far, and of course I have tested Rufus 3.1 on various Windows 10 systems, so I strongly suspect that this is something on your side.

1. Please open the exe with an editor (preferably an hex editor) and have a look at it. A lot of firewalls or security solutions seem to alter the downloads they want to filter and replace them with an HTML message, which of course means you get garbage that you can't run. Alternatively, please validate that the SHA-256 of the file you downloaded is: `A81CBA62D55EE42F41303C138A439D5DC420753A8AF12CB4753CF3DF67DCD067`

2. Can you indicate whether the Rufus 3.0 executable works? You can find it [here](https://rufus.ie/downloads/)?

3. You should download DebugView from [here](https://docs.microsoft.com/en-us/sysinternals/downloads/debugview) and run it as Administrator. Then in the _Capture_ menu make sure you capture _Win32_ and _Global Win32_. Then launch Rufus and see if you can see any output from Rufus (as it always sends some data there).

kaoscoach, 2018-06-20:
Looks like a Windows file in hexedit (first couple of lines):
00000000   4D 5A 90 00  03 00 00 00  04 00 00 00  FF FF 00 00  B8 00 00 00  00 00 00 00  MZ......................
00000018   40 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  @.......................
00000030   00 00 00 00  00 00 00 00  00 00 00 00  80 00 00 00  0E 1F BA 0E  00 B4 09 CD  ........................
00000048   21 B8 01 4C  CD 21 54 68  69 73 20 70  72 6F 67 72  61 6D 20 63  61 6E 6E 6F  !..L.!This program canno
00000060   74 20 62 65  20 72 75 6E  20 69 6E 20  44 4F 53 20  6D 6F 64 65  2E 0D 0D 0A  t be run in DOS mode....
00000078   24 00 00 00  00 00 00 00  50 45 00 00  4C 01 03 00  00 00 00 00  00 00 00 00  $.......PE..L...........

jrather@MM228963-PC ~/jrather/Downloads
$ openssl dgst -sha256 rufus-3.1.exe
SHA256(rufus-3.1.exe)= a81cba62d55ee42f41303c138a439d5dc420753a8af12cb4753cf3df67dcd067

3.0 version runs. 

I'll look into the DebugView.

kaoscoach, 2018-06-20:
So after a couple of other tests. I have found the problem. Apparently the issue is Chrome. If I download it with Chrome, even though the hash is the same, the file will not run. If I download it with IE all is well. Strangest thing I have ever seen and I really am unsure of how to explain it since the hashes are the same?

closed 2018-06-20 by kaoscoach

pbatard, 2018-06-20:
> even though the hash is the same, the file will not run

Wow, what the heck?!?

Considering that I've had reports of weird behaviour from Chrome (e.g. pbatard/libwdi#126) where it added unwanted extra payload to downloads, I'm starting to suspect that the Chrome developers are taking a bit too many liberties with their user experience these days (especially as I have recently been experiencing Chrome crashes where it managed to switched my monitor resolution to super low res - How on earth is a browser supposed to do just that?).

I suspect that Chrome probably tried to add Windows security related attributes on the download, and screwed that up, which in turn must have prevented the application launch or something.

I have however tested with Chrome on my side, and saw no issue, so I agree that this should probably be closed, as it seems a browser problem.