pbatard/rufus #1169: Windows Defender reports that Rufus 3.1 contains Trojan:Plutruption! or Trojan.Vigorf!

pbatard, 2018-06-23:
At least for `Plutruption`, this seems to occur with definitions `1.269.1834.0`.

Unfortunately, this is not the first time we've been having a false positive from Windows Defender (which also reported `Plutruption` for Rufus 3.0, but that was cleared as a __false positive__ in later definitions), and we have of course reported the issue (Submission ID: `b617cf91-4d99-409d-b5b4-4ebcba610237`)

However, if you have the feeling that Microsoft are not doing enough to avoid these false positives, or not acting quikcly enough, __you__ may also try to submit a false positive reportat https://www.microsoft.com/en-us/wdsi/filesubmission, to bring it to their attention.

pbatard, 2018-06-24:
Yeah, unfortunately it appears that someone, who wants to discredit our work, is "gaming" the Cloud reporting of Windows Defender, which in turns results in false positives in the definitions that Microsoft sends to Windows users.

For the time being, you can download the __updated__ 3.1 executable from the website, which is the same as the last one, but with newer timestamps (since Defender is obviously not looking at the actual behaviour of the executable to determine if it contains a trojan, but just trying to match the executable hash), and that shouldn't trigger Defender.

For the record, we have pre-emptively submitted that "new" executable to the defender team ([Submission ID: fc731445-c957-4bb3-b5cb-e388723116e6](https://www.microsoft.com/en-us/wdsi/submission/fc731445-c957-4bb3-b5cb-e388723116e6)) so that they aren't going to fall into the same trap...

plambrechtsen, 2018-06-24:
I had the same problem this morning and have also submitted it as a false positive.
https://www.microsoft.com/en-us/wdsi/submission/8583c667-5616-4b98-883f-18c4b461ae4e
Local version: 1.269.1861.0
Submitting it is still detected as malware:
Protection	Current detection			Definition version
Cloud		Trojan:Win32/Plutruption!ARXep	Online
Client		Trojan:Win32/Plutruption!ARXep	1.269.1901.0

pbatard, 2018-06-25:
Latest report for the original `.exe` (i.e. the one that was flagged as "malware"):

[image]

As expected, their __Final determination__ is of course that there is nothing malicious about the software, though they have yet to provide Analyst comments and update their definitions to clear out the __false positive__.

FaffeF, 2018-06-28:
FWIW, I got this alert in Windows Defender ATP, and the automated investigation service cleared it as benign inside an hour.

pbatard, 2018-06-30:
Since it's been a week, and Microsoft cleared the false positive, I'm going to close the issue. Note that the "old" executable has been reinstated as the default download on the website.

closed 2018-06-30 by pbatard