pbatard/rufus #1173: Rufus Build 3.1.1320 can't list / browse mapped drives (SMB shares)

airevolution, 2018-06-27:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus can't list / browse mapped drives (SMB shares)
	Others drives show alright in the SECTION Window, but not the letters from my NAS and its shares are mapped with a letter and propoerly connected to Windows, since I ran rufus from one of those shares...   Rufus 2.8 worked fine Under same conditions

Log
---
Rufus Build 3.1.1320
Windows 7 pro 32 bits / Rufus was ran as Elevated Privilege when prompted to
```

airevolution, 2018-06-27:
Of course, I tried to reboot or close/open new Windows session

Share work fine with everything else
Even previous version of Rufus (2.8)

Thank's

airevolution, 2018-06-27:
Correction : Previous version were 2.6p and 2.18

On the problem :  Found the cause and a work around.
It isn't Rufus directly, but for the unrecognised digital signature of the software, Rufus rise a Windows Users Control alert, which of course you have to answer by yes if you want Rufus to run.  And Windows,  the bugger, ran it as a different user or under a different session somehow (not clear here since it's the first time Windows behave like this (no mapped drives) when i'm logged in with admin rights... event when you do a SHIFT + RIGHT CLIC to run as another account...  those command, brough up with them the mapped drives in the past).
So in the same browse Window, I did a right clic on Computer and connected the Mapped drive...
And now it works fine !

Maybe check the digital signature of Rufus so it get a cert from a free and well known cert authority.

pbatard, 2018-06-27:
Answered in the FAQ: https://github.com/pbatard/rufus/wiki/FAQ#Why_can_Rufus_not_see_my_network_shares

The digital signature is from a well know cert authority (else Windows would not let you run the application at all). And of course you will get a UAC prompt when running Rufus, since it needs to run as Administrator.

We did however __renew__ our certificate between 2.18 and 3.0, since you're not going to find anybody providing Authenticode credentials that don't expire, and I am pretty sure that you must have fiddled with your "always allow software signed by this certificate" Windows setting with 2.x, and didn't do the same with 3.x.

At any rate, there's nothing for me to do here, since I cannot __revert__ to a certificate that has __expired__ just so you can avoid the prompt. You will have to sort your issue on your own (preferably by applying the workaround indicated in the FAQ entry above).

closed 2018-06-27 by pbatard