pbatard/rufus #1175: supper talent express rc4 not supported

zabdulla1, 2018-06-29:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
rufus cannot detect the drive, even though windows can detect it and can make win2go bootable drive.
[rufus.log](https://github.com/pbatard/rufus/files/2147421/rufus.log)

Log
---
```
<FULL LOG>
```

pbatard, 2018-06-29:
Please provide a new log after pressing <kbd>Alt</kbd>-<kbd>.</kbd> in Rufus to enable USB enumeration debugging. You should find that your log is much longer and contains information about device detection.

pbatard, 2018-06-29:
Note: You may still be able to access it using the undocumented and unsupported <kbd>Ctrl</kbd>-<kbd>Alt</kbd>-<kbd>F</kbd> cheat mode in Rufus, but if you use that cheat mode, you're on your own.