pbatard/rufus #1176: HDD not detected if in a box USB\VID_152D&PID_0578

F5BJR, 2018-07-04:
As in the tittle: HDD not detected if in a box USB\VID_152D&PID_0578

https://devicehunt.com/view/type/usb/vendor/152D/device/0578

﻿

pbatard, 2018-07-04:
Didn't follow the checklist &rarr; Closed.

closed 2018-07-04 by pbatard

F5BJR, 2018-07-04:
Found : ALT + F and HDD is detected ...

pbatard, 2018-07-04:
Again, the checklist __explicitly__ advised you to read the FAQ, where you would have easily found the answer to your question, under the topic: _Help, my USB drive is not detected!_

I am not asking you to follow the checklist to be annoying, but because if you followed the advice there, you would have easily found the answer to your question. The checklist exists for a reason: to help __YOU__!!!

F5BJR, 2018-07-04:
Sorry for disturb but since i used Rufus I did not have this problem
I may be silly but if the information was indicated in the GUI, it would have avoided asking the question

pbatard, 2018-07-04:
The GUI is designed to hide USB HDDs by default (the reason for that is another question that is answered in the FAQ). So that's not going to change.

Besides, if you do want to get a hint about what to do to get your USB HDD listed in the GUI, you will actually get it in the log.

If Rufus doesn't behave as you expect you should check the log (which is pretty much what the checklist __insists__ that you do). Thus, if you have followed the checklist, you would have seen something similar to this in your log:
```
Found UAS (USB 3.0) device 'Seagate Expansion Desk SCSI Disk Device' (0BC2:3322)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
```
The last line would have told exactly what you needed to do if to see your drive.

So you saw the rules, but decided that, somehow, you knew better and that they should not apply to you. That is entirely your problem. So don't try to twist it and pretend that the fault lies with the application.

F5BJR, 2018-07-04:
For the sugestion it is to add if it is possible at the bottom of the screen next to : 
 0 devices detected the text: use ALT + F for force hdd detection

I think it's not as stupid as that

pbatard, 2018-07-04:
I will never advertise cheat modes in the UI, and <kbd>Alt</kbd>-<kbd>F</kbd> is a cheat mode. They are hidden for a reason, just like the drives.

If I was making it too explicit to list USB HDDs, then I'd start receiving plenty of complaints from people who formatted a backup HDD by mistake, and lost valuable data. Even with hiding HDDs by default, I already get too many reports from people who mistakenly erased the wrong drive. So I will __not__ advertise explicitly how to list them in the UI.

As I say in the FAQ:
> I'd much rather inconvenience a few people, by not letting them (easily) erase the data they want, than inconvenience others by allowing them to erase data they don't want to erase.

F5BJR, 2018-07-04:
c'est stupide supprimez donc tout ...

pbatard, 2018-07-04:
Next time, if you see instructions with a big __PLEASE DON'T IGNORE THIS AND FOLLOW THESE STEPS__, please don't think that, for whatever reason, that request should not apply to you. It will avoid __everyone__ having to waste time on "stupid" questions that are answered at length in the FAQ... including yourself.