pbatard/rufus #1178: Issue with booting: Lauching NTFS EFI loader 'efi\boot\bootx64 efi

dolarcher, 2018-07-07:
Hi guys,
I am new here and have problem with booting microsoft software (WIN 8.1) to my computer.
Settung in rufus: I have Sandisk USB with 32 GB; took original ISO from microsoft page x64; choose in Rufus UEFI and NTFS. 
Now when I stick USB to my computer first it was not booting usb right then I made setting in boot menu: Lauch CSM enable ; Secure Boot Controle disable.
After this seetings usb loading but now I got the following message (see screen). After that computer show blue screen.

pbatard, 2018-07-07:
Sorry, but you'll have to recreate this issue after:
1. Following the checklist, and __ESPECIALLY__ providing the log of how you created your USB with Rufus. It doesn't matter if your error is when booting, I need to see how you created the USB, or I can't help you.
2. Finding a way to include the screenshot if you mention a screenshot.

The checklist is __NOT__ optional. If you don't follow it, then your issue gets closed.

pbatard, 2018-07-07:
Okay, I see you managed to attach the screenshot so I'm unlocking and reopening this issue. But please provide your log. You may have to recreate your USB to do so.

pbatard, 2018-07-07:
Your ISO doesn't have a >4GB file, so there's no need to use NTFS. Is there a specific reason why you changed the file system to NTFS instead of FAT32?

What happens if you create your USB using FAT32?

Also, please compute the SHA-1 of your ISO (which you can do in Rufus by clicking the  `(✓)` button, and which the checklist you __chose__ to ignore advises you to do) and copy/paste the value here.

dolarcher, 2018-07-07:
The ISO from microsoft WIN8.1 x64 has 4.1GB - should I try now with FAT32?
MD5: f4a9c14daa0f2e205e6c025f6db07a8b
SHA1: 07c464680da474a4d6c23d57d99a90f129f85584
SHA256: 29be9b2f13363c935cbfc967d2f7cf095b6ac18727a327345ce09b276a2034d2

pbatard, 2018-07-07:
> The ISO from microsoft WIN8.1 x64 has 4.1GB

No, the ISO __is__ 4.1 GB, not "has". But what matters is if it __contains__ a file that is > 4GB. If you have an 8 GB ISO with 8 x 1 GB files, you can use FAT32. It's only if the ISO __contains__ a file that is larger than you have an issue. But the largest file on your ISO is 3.5 GB, so you can use FAT32.

> should I try now with FAT32?

That's what I asked you to do. __Of course__ if I tell you to use FAT32, it's because I have checked that you can use FAT32. Also, Rufus is smart enough to detect if you can use FAT32 or not, and remove it from the list if not. If Rufus allows you to select FAT32, the you __CAN__ use FAT32. Please have a little more faith in what people or applications tell you, and don't try to second guess.

pbatard, 2018-07-07:
PS: Your SHA-1 looks correct for `de_windows_8.1_with_update_x64_dvd_6051485.iso`, so I can tell that you are indeed using an official Microsoft ISO and that it is not corrupted, which is definitely something that you want to validate if you encounter an issue during boot.

dolarcher, 2018-07-07:
I also tried to use same USB with Ubuntu (FAT32 was selected there) - there I had no problems and my MEDION computer worked with that.