pbatard/rufus #1179: Extremely slow Windows To Go USB with Windows 10 Pro

skoushan, 2018-07-13:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Trying to create a WIndows To GO USB for Windows 10 Pro. The "Applying Windows image" step is taking approximately 10 minutes per 0.1% (i.e. it's going to take a week at this rate). From the log, it's mostly taking time to extract small files like PNG's, though CPU usage from Rufus is below 1%. My device is USB 3.0 (and Rufus recognized that). What could be the problem here?

Log
---
```
1 device found
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3766, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x8A3FABF0
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.8 GB (30975000576 bytes)
  Start Sector: 8064, Boot: No
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Size: 4.4 GB (Projected)
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: Win10_1803_English_x64.iso (4.4 GB)
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM0\sources\install.wim:[0] (API)
  Could not access image: [0x0000000D] The data is invalid.
Failed to acquire WIM index
Found USB 3.0 device 'TOSHIBA USB FLASH DRIVE USB Device' (0930:6545)
1 device found
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3766, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x8A3FABF0
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.8 GB (30975000576 bytes)
  Start Sector: 8064, Boot: No
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM0\sources\install.wim:[0] (API)
  Could not access image: [0x0000000D] The data is invalid.
Failed to acquire WIM index
Found USB 3.0 device 'TOSHIBA USB FLASH DRIVE USB Device' (0930:6545)
1 device found
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3766, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x8A3FABF0
[... 1512 lines cut ...]
Extracting: E:\Program Files\WindowsApps\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\28.png (623 bytes)
Extracting: E:\Windows\InfusedApps\Packages\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\27.png (680 bytes)
Extracting: E:\Program Files\WindowsApps\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\27.png (680 bytes)
Extracting: E:\Windows\InfusedApps\Packages\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\3.png (444 bytes)
Extracting: E:\Program Files\WindowsApps\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\3.png (444 bytes)
Extracting: E:\Windows\InfusedApps\Packages\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\30.png (403 bytes)
Extracting: E:\Program Files\WindowsApps\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\30.png (403 bytes)
Extracting: E:\Windows\InfusedApps\Packages\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\50.png (733 bytes)
Extracting: E:\Program Files\WindowsApps\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\50.png (733 bytes)
Extracting: E:\Windows\InfusedApps\Packages\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\5.png (336 bytes)
Extracting: E:\Program Files\WindowsApps\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\5.png (336 bytes)
Extracting: E:\Windows\InfusedApps\Packages\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\55.png (377 bytes)
Extracting: E:\Program Files\WindowsApps\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\55.png (377 bytes)
Extracting: E:\Windows\InfusedApps\Packages\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\6.png (481 bytes)
Extracting: E:\Program Files\WindowsApps\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\6.png (481 bytes)
Extracting: E:\Windows\InfusedApps\Packages\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\63.png (135 bytes)
Extracting: E:\Program Files\WindowsApps\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\63.png (135 bytes)
Extracting: E:\Windows\InfusedApps\Packages\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\resources.pri (743 KB)
Extracting: E:\Program Files\WindowsApps\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\resources.pri (743 KB)
Extracting: E:\Windows\InfusedApps\Packages\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\66.png (767 bytes)
Extracting: E:\Program Files\WindowsApps\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\66.png (767 bytes)
Extracting: E:\Windows\InfusedApps\Packages\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\7.png (586 bytes)
Extracting: E:\Program Files\WindowsApps\Microsoft.BingWeather_4.22.3254.0_x64__8wekyb3d8bbwe\Assets\AppTiles\WeatherIcons\30x30\7.png (586 bytes)
// CONTINUED
```

pbatard, 2018-07-13:
It doesn't matter if your device is USB 3.0 if the flash memory inside is slow. Alas, it's not at all uncommon for cheap USB 3.0 device to use flash that can only write at 10 MB/s, in which case you are going to find that applying a Windows image can be excruciatingly slow.

Unless your device can sustain 60 MB/s __write__ speed (be mindful that a USB 3.0 device might be rated 100MB/s read speed, but still have a 10 MB/s write speed), you should consider using a different device for Windows To Go, as you will only be disappointed by both the speed at which you can create the Windows To Go drive, and the speed at which it operates.

Also, because Windows is basically using the Windows APIs to apply an image, which means that it's a complete black box, there's literally nothing we can do to speed up things, as Microsoft are the only ones who control what's going on behind the scenes here.

For the record, with a USB 3.0 flash drive that has a 70 MB/s write speed, creating a Windows To Go drive takes less than 15 mins.

skoushan, 2018-07-13:
Got it, thanks. Will try with a different stick and re-open if it's still problematic.

closed 2018-07-13 by skoushan

skoushan, 2018-07-16:
Tried it with a SanDisk CZ80; done after about 30 minutes. Thanks for building such a great tool!

pbatard, 2018-07-16:
Alrighty then. Of course, creation times may also vary depending on the image you use (I think more recent Windows 10 images are slower than the earlier Windows 10 ISOs), but 30 mins sounds about right, and should leave you with a Windows To Drive that is actually usable. 😃

Thanks for following up on this issue and posting an update!

ibuprophen1, 2018-07-16:
Just a thought based on my experience... 

I'm not sure if this applies to your specific system but, i recall that the speed on my USB 3.0 port had dramatically improved in R/W speed by updating the Drivers from the Chipset Manufacturer and not the Motherboard/PCI-E Card Manufacturer because, in my case, the Motherboard/PCI-E Manufacturers drivers were fine but, the Manufacturer for the Chipset was so much newer and performed so much better. 

In addition to this, I've also found that on Windows 8.1/10, the Windows Indexer always has been affecting this and other areas of the R/W for drives (including this USB Process in question). 

At times I've had to go and end the Indexing Process to noticeably increase the speed for writing to an external drive. 

Just my thoughts that I hope is helpful to you! :-) 

~Ibuprophen

pbatard, 2018-07-16:
That's good to know. For the record, recent version of Rufus do disable drive indexing when formatting the partition, which should hopefully keep Windows away from trying to poke around the files being copied over. But then again, Rufus also holds an exclusive lock on the partition while writing files to it, which, even with indexing enabled, should keep Windows at bay. But of course, one can never tell when Microsoft might have decided to have Windows ignore rules, that applications try to set to keep it at bay...