pbatard/rufus #1180: RUFUS does not create a UEFI Bootable USB that works on DELL XPS 93xx models

Lucero7919, 2018-07-19:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Tried RUFUS 2.1 and v3 and neither create a UEFI Boot USB that works in DELL XPS 93xx models using Windows 10 1803 for business editions ISO (from MSDN)

Created using GPT UEFI (non CSM)
Tried with secure boot enabled and receive a Start Failre 17 No Mapping error
Tried with secure boot disabled (not really acceptable in this day and age) and receive the digital signing error associated with secure boot.

When booting to USB, boot options only show the USB stick with partition 2 (I assume RUFUS creates it's own partition to boot UEFI that the Boot loader cannot see). 

Works fine if I set everything to Legacy BIOS

Log attached
[rufus.log](https://github.com/pbatard/rufus/files/2209141/rufus.log)

pbatard, 2018-07-19:
So, as indicated by the log, the image you use has a >4GB file (`install.wim`), which means you can't use FAT32.

As a result NTFS must be used and, to boot in UEFI mode, Rufus must create an UEFI:NTFS partition (see [here](https://github.com/pbatard/uefi-ntfs)) that first loads an UEFI NTFS driver, and then hands over the boot to the Microsoft bootloader on the NTFS partition. Your problem appears to be with actually running the UEFI:NTFS bootloader (which isn't signed for Secure Boot but more about this below).

> Tried with secure boot disabled (not really acceptable in this day and age) and receive the digital signing error associated with secure boot.

Two things:

1. If you receive a _"digital signing error associated with secure boot"_, then clearly, you have __not__ disabled Secure Boot. If Secure Boot is disabled, then it shouldn't complain about the boot file failing the Secure Boot validation. So please check your settings, because it definitely looks like you didn't actually do what you mentioned you did. If you retry after making sure that Secure Boot is effectively disabled, should find that things work a lot better.

2. _"not really acceptable in this day and age"_ I beg to differ when it's __temporarily__ for the initial installation boot process. If you do understand what Secure Boot is about, and what it is meant to do, then, if you validated the SHA-1 of the ISO you downloaded to ensure that it is the official one from Microsoft, and used a (hopefully trustworthy) digitally signed application to create your USB flash drive, then you have pretty much carried the same steps as Secure Boot would do in __validating__ that the boot files you are going to use have not been tampered with. This is really __ALL__ Secure Boot does, and, as such, even if Secure Boot is disabled, that doesn't mean there doesn't exist another way of accomplishing the same task. Please see the second part [of this](https://github.com/pbatard/rufus/wiki/FAQ#Blah_UEFI_Blah_FAT32_therefore_Rufus_should_Blah). This means that, despite what many people seem to believe, it is perfectly acceptable to __temporarily__ disable Secure Boot during the initial phase of your OS installation, if you have validated that your OS media comes from official sources and that it was created in a trustworthy manner.

Now, if you are really unhappy about having to disable Secure Boot to boot a media that contains a >4GB file, such as the official updated Windows 10 ISOs from Microsoft, then __PLEASE COMPLAIN TO MICROSOFT ABOUT THEIR BLATANT ABUSE OF POWER IN REFUSING TO SIGN ANYTHING THAT IS GPLv3 FOR SECURE BOOT__!

You see, __I would like nothing more than to sign the UEFI:NTFS bootloader for Secure Boot__, to keep people who have an irrational fear of even temporarily disabling it happy.

But I can't, because UEFI:NTFS is GPLv3 (the reason for that is that the only Open Source NTFS driver I know, which comes from GRUB, of is GPLv3, and rewriting a read-only UEFI NTFS driver from scratch, and licensing it under terms that provide __LESS__ freedom to users, just because Microsoft have decided to blatantly abuse their power, doesn't seem like a good endeavour), and Microsoft, which is the __only__ entity that controls the Secure Boot signing process, has a clause in the Secure Boot signing conditions that __explicitly forbids anything GPLv3 from ever being signed for Secure Boot__ (under the flimsy pretext that it would force them to relinquish their private signing keys, which of course is as ridiculous as it gets when you read and understand the intent of the GPLv3)... If that is not blatant abuse of power, against a license that they have long stood against, then I don't know what is.

So, you have 3 things there:
- Figure out what you did wrong when disabling Secure Boot, because, according to what you report, it certainly didn't look like it was disabled at all to me...
- Accept that temporarily disabling Secure Boot for the initial booting of the media is not a big deal, when you have validated that the media you used for your USB Flash Drive creation is trustworthy
- Instead of complaining to the maker of an application, who would like nothing better than have it signed for Secure Boot, that it doesn't work with Secure Boot, recognize when a blatant abuse of power is at play, and complain to the entity that is abusing its power instead (here Microsoft).

pbatard, 2018-07-19:
> When I disabled, I received the error listed above "Start Failure 17 No Mapping error"

Oh, did you see the `UEFI:NTFS` start message then?
Can you please list them all (a screenshot will do if you don't want to have to type them).
I just want to be sure I have the whole sequence before I investigate further...

Lucero7919, 2018-07-19:
Yes. I'll grab a shot for you. Probably about 10 lines in total. 

I pulled just the index I wanted from the iso which meant I could create the USB as Fat32 but still didn't work. I'll post the screenshot when I'm home

pbatard, 2018-07-23:
Any update on this? I'd still like to see the screenshot...

pbatard, 2018-07-31:
No update &rarr; Closed.

closed 2018-07-31 by pbatard