pbatard/rufus #1183: Win 7 IOS Formating to FAT32 with RUFUS but not formating right (NTFS option is only)

Meliodas15, 2018-07-21:
﻿

Checklist
---------
- [X ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ]X I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X ] I also tried one or more of the following:
  - [ X] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ X] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<So i never done USB FLAsh DRIVE Installing windows and i am trying whole day without succses
Problem is that my pc doesn't recognise damn NTFS and needs Fat32 but rufus doesn't do a thing about this still formating at NTFS But i really need FAT32 (win 7 Iso) so is any way to help me with this? when i try to boot with USB FLASH DRIVE it doesn't even recognise and starts normal windows without setup or something i did also in bios settings to recognise but still deosn't even pop-up>

Log
---
```
<FULL LOG>
```

pbatard, 2018-07-21:
> But i really need FAT32 

Why?

If your computer is BIOS based, FAT32 or NTFS for bootable __installation__ media doesn't matter one bit, and, in fact, NTFS is by far preferable, because it allows to support >4GB `install.wim` images, which FAT32 cannot do. This is the very reason why Rufus doesn't offer you the option to select FAT32 for BIOS based Windows installation, because, I have yet to see the hint of a single demonstrable issue with using NTFS over FAT32 for BIOS.

And if your computer is UEFI based, then (after you make sure you select UEFI in _Target type_), Rufus should allow you to select either FAT32 or NTFS, as long as you don't have a >4GB `install.wim` (because otherwise, FAT32 can simply __NOT__ be used as it can't accomodate a >4GB file). But even then, despite what many people believe, using NTFS over FAT32 for UEFI boot is [__not that big of an issue__](https://github.com/pbatard/rufus/wiki/FAQ#Blah_UEFI_Blah_FAT32_therefore_Rufus_should_Blah).

Now, because you __chose__ not to provide your log, despite being explicitly asked to do so when opening the issue, I have no idea whether your ISO contains a >4GB file (but if it's an official Windows 7 image from Microsoft, I'm pretty sure it doesn't), as well as other details that could really help with the troubleshooting. So the only thing I can advise you is:
1. Instead of trying to play with the file system, which doesn't matter, play with the _Partition scheme_ and _Target system_ options instead
2. Look into your BIOS or UEFI settings. If using UEFI, you may have to (temporarily) disable Secure Boot, or play with the CSM option for your USB media to boot.

pbatard, 2018-07-23:
No update &rarr; Closed.

closed 2018-07-23 by pbatard

Meliodas15, 2018-07-24:
I found solution in Bios settings needed to change one thing like first boot needed to be USB FLASH Drive and i was changing wrong thing in BIOS then i realised and worked fine thanks anyway.