pbatard/rufus #1185: Rufus 3.1.1320 fails to download additional files with error 0x00002f19 (12057)

kevinjwalters, 2018-07-24:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Fedora 28 workstation iso will not write on a vanilla Windows 10 host. Error dialogue pops up: **Unknown internet error 0x00002f19**. I mentioned this in comments on #1045

I'm not very au fait with Windows. Does this error relate to a TLS issue with https?

Log
---
```
Rufus version: 3.1.1320
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found card reader device 'Generic SD16G SD Card'
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1911, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x3A663A44
Drive has an unknown Master Boot Record
Partition 2:
Type: EFI FAT (0xef)
Size: 9.1 MB (9551872 bytes)
Start Sector: 119940, Boot: No
Scanning image...
ISO analysis:
Image is an ISO9660 image
Will use '/isolinux/isolinux.cfg' for Syslinux
Detected Syslinux version: 6.04 (from '/isolinux/isolinux.bin')
Disk image analysis:
Image has an unknown Master Boot Record
Image is a bootable disk image
ISO label: 'Fedora-WS-Live-28-1-1'
Size: 1.7 GB (Projected)
Note: File on disk is larger than reported ISO size by 896 KB...
Uses: Syslinux/Isolinux v6.04
Uses: EFI
Using image: Fedora-Workstation-Live-x86_64-28-1.1.iso (1.7 GB)
Downloading 'ldlinux.sys' from https://rufus.ie/files/syslinux-6.04/ldlinux.sys
Unable to send request: Unknown internet error 0x00002F19
Could not download the file - cancelling
```

pbatard, 2018-07-24:
The https://rufus.ie/files/syslinux-6.04/ldlinux.sys link is valid and I am seeing no issue downloading that file.

Internet error `0x00002F19` is `ERROR_INTERNET_SEC_CERT_REV_FAILED` which means your computer was not able to complete the certificate revocation check (which __is__ completely different from meaning that the certificate using by the server has been revoked).
In other words, the problem is 100% on your end. One possibility is that your clock is not set to the current date & time (revocation validation is very time sensitive), or you have disabled revocation checks altogether.

At any rate, you will have to solve that error yourself, as it is purely environmental and does not come from  https://rufus.ie.

(NB: Since you opened a new issue, I will delete your comments from the one you piggy-backed on)

JonnyTech, 2018-07-24:
Is the computer correctly connected to the internet? Do you have a firewall blocking connections?

kevinjwalters, 2018-07-24:
Thanks for your fast response. My time is correct. I have made no changes to revocation checks. For comparison, I can access same file in Chrome, Firefox and Edge (the Microsoft browser, not the U2 lead guitarist).

Comment to all: would be interested to hear if any other Windows 10 users have the same issue.

kevinjwalters, 2018-07-24:
This may be my ISP irrationally blocking the OCSP service. This was the last thing I expected! Let me do some more digging...

pbatard, 2018-07-24:
Try to access it with IE (Internet Explorer), which is still included in Windows 10. This uses the same APIs as the one Rufus uses to download the files.
Just open your Start menu and type `iexplore` then copy/paste the https://rufus.ie/files/syslinux-6.04/ldlinux.sys link.

I'm half convinced you will get the same issue then.

For the record, no, I am not seeing the same issue with IE on my machine, and, considering that none of the 3 million people who download Rufus every month are reporting a similar issue, I am confident it is purely environmental on your side.

kevinjwalters, 2018-07-24:
I have reported the mis-categorisation of http://ocsp.int-x3.letsencrypt.org to my ISP. I assume they have a data provider for categories so this could affect more than other users of my large, UK-based ISP. For comparison, I had a quick glimpse at Websense's data and they have it in a correct, reasonable category. 

IE works. Possible the browser is more permissive on CRL/OSCP to cope with transient problems and limited/filtered Internet connectivity or has an alternate privacy enhanced scheme, cf Chrome's CRLsets.

I note Powershell's Invoke-WebRequest works ok. Can I just download the files with this and place them somewhere as a workaround until my ISP's proxy category data is corrected?

pbatard, 2018-07-24:
You'll need to copy the 6.04 `ldlinux.sys` and `ldlinux.bss` into `rufus_files\syslinux-6.04\`.

kevinjwalters, 2018-07-26:
Thanks, that works as a work around. The category data has been corrected and my ISP now no longer blocks the OCSP responder. By renaming the directory, I verified that rufus can now download files without the 12057 error.

That was certainly a strange one!

closed 2018-07-26 by kevinjwalters