pbatard/rufus #1187: Flash memory is dead "x86 Master Boot Record"

mahmoodn, 2018-07-30:
With Rufus 2.6, I wanted to create some bootable devices on a 16GB flash drive. While trying to test GPT and MBR, I got a write error during the format and now the flash memory is completely unusable! 

My computer doesn't show that. In disk management, I can see the device, but I am not able to format it. Even I tried to delete partition and create a new one.

The flash is dead! Rufus 2.6 says
```

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
No drive letter was assigned...
Will use 'H:' as volume mountpoint
Could not get Disk Extents: [0x00000001] Incorrect function.
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error [0x00000005] Access is denied.
  Wrote: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
unable to zero MBR/GPT
Could not get Disk Extents: [0x00000001] Incorrect function.

Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (8564:1000)
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
Found USB 2.0 device 'WD Elements 10A8 USB Device' (1058:10A8)
NOTE: This device is an USB 3.0 device operating at lower speed...
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
Device eliminated because it was detected as an USB Hard Drive (score 23 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x82
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000001
Drive does not have an x86 Master Boot Record
Partition 1:
  Type: Small FAT16 (0x04)
  Size: 15.1 GB (16231956480 bytes)
  Start Sector: 0, Boot: No, Recognized: Yes

```

I also tried Rufus 3.1 which says 

```
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE2 for exclusive write access
Requesting lock...
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
No drive letter was assigned...
Will use 'H:' as volume mountpoint
Could not get Disk Extents: [0x00000001] Incorrect function.
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not reset partitions
Could not get Disk Extents: [0x00000001] Incorrect function.

Found non-USB removable device 'WDC WD10JPVX-08JC3T5' => Eliminated
Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (8564:1000)
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
Found USB 2.0 device 'WD Elements 10A8 USB Device' (1058:10A8)
NOTE: This device is an USB 3.0 device operating at lower speed...
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x82
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1973, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 15.1 GB (16231956480 bytes)
  Start Sector: 0, Boot: No

What should I d
```o? I see similar issues in the repository but I don't see any clear solution for that.

ibuprophen1, 2018-07-30:
If you're referring to the "JetFlash Transcend 16GB USB Device", see if the following JetFlash Repair tool (from the manufacturer) works. 

https://www.transcend-info.com/Support/Software-3/ 

Good Luck! 

~Ibuprophen

pbatard, 2018-07-31:
There's nothing I can do about hardware failures that coincidentally happen while you are using Rufus.
If your failure had happened while you were copying files while using Windows Explorer, you wouldn't be contacting Microsoft support to ask them how you can fix your hardware problem.

And, yes, despite what a small minority is trying to pretend (yet, despite years of trying to do so, unable to prove while it should be __exceedingly easy to do__, if it was really a software issue), __unrelated__ hardware failures can happen while you are using software, and even repeatedly, if you happen to use cheap flash drives that report a "fake" capacity, which, unfortunately, unscrupulous sellers do appear to have plenty of. But that still doesn't mean that the software makes can do anything to help with what is essentially a hardware failure.

If you are seeing repeated errors, have tried to check for bad blocks on multiple machines (preferably with different OSes), and still can't seem to recover your drive, then your hardware is faulty and you need to purchase a new drive. That's all.

closed 2018-07-31 by pbatard

mahmoodn, 2018-07-31:
@ibuprophen1
The tool says no device has been found!

The flash isn't a cheap one and I have used for several years. I doubt if there is a hardware failure on that because I used the flash right before attempting to create boot device.

It seems that MBR is erased because I set GPT in one of my attempts. The question is how can I manually create a MBR partition? With Linux command line? Other tools?