pbatard/rufus #1188: RUFUS causes Blue Screen Error APC_INDEX_MISMATCH while trying to install FreeNAS

CyberWeather, 2018-08-05:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
While trying to put FreeNAS 11.2 BETA 2 on a SanDisk Cruzer Blad 8GB Flash Drive gives me a Blue Screen error called APC_INDEX_MISMATCH, while I have factory reset my system, the problem still persists on _ONLY_ this iso. I've tested CentOS and XCP-ng on the same USB Drive. Below is a log tested on a different computer and the log has the following output. Note I couldn't retrive the log for the system it crashed. I've also taken the issue to Microsoft Community which can be found [here](https://answers.microsoft.com/en-us/windows/forum/windows_7-hardware/blue-screen-apcindexmismatch-0x0000001-using-rufus/3b8c5124-632a-4233-9aa1-b3748b03c0aa?) to see if this issue can be resolved.

Any Thoughts?

Log
---
```
Rufus version: 3.1.1320 (Portable)
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 954, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has an unknown Master Boot Record
Partition 1:
  Type: GPT Protective MBR (0xee)
  Size: 639.7 MB (670780928 bytes)
  Start Sector: 1, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'FREENAS'
  Size: 636.2 MB (Projected)
  Note: File on disk is larger than reported ISO size by 300 KB...
  Has a >64 chars filename
This ISOHybrid is not compatible with any of the ISO boot methods we support
Using image: FreeNAS-11.2-BETA2.iso (639.7 MB)

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'D:' as volume mountpoint
Requesting lock...
Bad Blocks: Checking from block 0 to 15330303
Bad Blocks: Using offset 443 for fake device check
Bad Blocks: Writing test pattern 0xAA
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Writing Image...
Drive has an unknown Master Boot Record

Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 954, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has an unknown Master Boot Record
Partition 1:
  Type: GPT Protective MBR (0xee)
  Size: 639.7 MB (670780928 bytes)
  Start Sector: 1, Boot: No
```

pbatard, 2018-08-05:
I'll try to test this when I get a chance, but can you please test what happens when you win32diskimager on Windows to write that image?

I strongly doubt this has anything to do with Rufus and even less with something that we can fix as, as the image writing process is exceedingly basic and straightforward: write every block of data as is.

If Windows parses the data we send to the drive (could be a Windows Defender issue), and then crashes, whereas all we are doing, really, is send it sequentially, there really doesn't exist much of anything to fix.

Also, is there a reason why you used the `.iso` (which, because it has to be crafted as a __hack__ of 2 incompatible file systems to work as an image can be problematic) instead of the `.txz` (which I assume is a proper DD image)? I think the `.txz` should be supported by Rufus, but I haven't tested it yet. What happens if you try with the `.txz`?

I'll see whether I can replicate the issue once I have downloaded that image.
For good measure, can you please confirm that the SHA-256 of that image is `b7a628801541c111f1e56a7810bb180479d9c156540e053db2b7f8cba797d43c`?

pbatard, 2018-08-05:
Tested on Windows 10 x64, and saw no issue:
```
Rufus version: 3.1.1320
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0809
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
No volume information for drive 0x85
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC765BBD0
Drive has an unknown Master Boot Record
Partition 1:
  Type: GPT Protective MBR (0xee)
  Size: 639.7 MB (670780928 bytes)
  Start Sector: 1, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'FREENAS'
  Size: 636.2 MB (Projected)
  Note: File on disk is larger than reported ISO size by 300 KB...
  Has a >64 chars filename
This ISOHybrid is not compatible with any of the ISO boot methods we support
Using image: FreeNAS-11.2-BETA2.iso (639.7 MB)

Computing checksum for 'D:\ISOs\FreeNAS-11.2-BETA2.iso'...
  MD5:	 d042eec458f92bfe325905ecccecc3eb
  SHA1:	 328553b26761012ebd2fcf87a607c33fc03d43c9
  SHA256: b7a628801541c111f1e56a7810bb180479d9c156540e053db2b7f8cba797d43c

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE5 for exclusive write access
Requesting lock...
No drive letter was assigned...
Will use 'G:' as volume mountpoint
Requesting lock...
Writing Image...
Drive has an unknown Master Boot Record

Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
No volume information for drive 0x85
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC765B81D
Drive has an unknown Master Boot Record
Partition 1:
  Type: GPT Protective MBR (0xee)
  Size: 639.7 MB (670780928 bytes)
  Start Sector: 1, Boot: No
```

I'm not sure I have a build 14393 of Windows 10  64-bit anywhere to test. Possibly this was an OS bug that Microsoft fixed in subsequent Windows 10 updates...

CyberWeather, 2018-08-05:
Hello thank you for your reply,

I forgot to mention the system affected is a Windows 7 Pro x64. I also tested on Windows 10 x64 on a laptop and got the same if not very similar result as you. 

I just did the standard "find the download and flash the iso" and frankly did not find out about a ".txz" file. So ill try it when I get a chance.

And Yes I already confirmed the sha256 hash using rufus.

pbatard, 2018-08-05:
Please try win32diskimager or any other `dd`-like application on that Windows 7 system, so that we can confirm that this is not related to using Rufus. I'll try to test on a Windows 7 machine too when I get a chance.

pbatard, 2018-08-05:
Also, another thing you can try: Can you try writing to a VHD? See [here](https://github.com/pbatard/rufus/wiki/FAQ#How_do_I_create_a_VHD_drive_to_use_with_Rufus) for how you can create a VHD to use with Rufus. This would tell us if your issue has anything to do with sending data over the USB bus.

ibuprophen1, 2018-08-05:
It's been quite a while since I ever saw the BSOD... 

Did you see if Refreshing the USB Stack Drivers resolves your BSOD issue? 

If your interested, I attached a txt file (to print as well if needed) with some simple (hopefully understandable) steps on how to do this. 

[Refreshing-USB-Stack-Driver.txt](https://github.com/pbatard/rufus/files/2260549/Refreshing-USB-Stack-Driver.txt)

Good Luck! 

 ~Ibuprophen

CyberWeather, 2018-08-05:
Not seening any at the moment.

ibuprophen1, 2018-08-05:
Did you try to obtain any updated drivers from the manufacturer of the USB/Chipset? 

~Ibuprophen

pbatard, 2018-08-05:
@ibuprophen1, I'd like to remind you that I too am seeing the BSOD on a __vanilla__ Windows 7 system, with only USB 2.0 (hence using native Microsoft USB drivers), and, from what I suspect, the issue most likely has to do with Windows trying to parse a partition table that it doesn't know how to properly process, during a kernel-mode operation. This is unlikely to have anything to do with the USB stack.

I've seen Windows (10 for that matter) declare an USB Flash Drive as completely inaccessible, no matter what you try (reboot, etc.) on account of a GPT table being set in a manner that Microsoft probably didn't expect (whereas Linux had no issue with the same drive, and, unlike Windows, could repartition/reformat it without any issue), so I have some doubts about the ability of Windows to properly handle all cases of partitioning. Instead, it would seem to me like Microsoft are patching their poor handling of partition tables as they go along, which would explain why Windows 7 crashes.

At any rate, if you want to take the matter further up, and considering that we have established that this has __nothing__ to do with Rufus, I will encourage you to do so on a Microsoft forum, as this is a pure Windows problem.

As such, I will close this issue.

closed 2018-08-05 by pbatard

ibuprophen1, 2018-08-05:
@pbatard, understood!

I too had seen this about a year ago +/- on the same Windows 7 build and had cleared it up within a few days involving both the GPT and MBR as well as an incompatible driver that Windows had defaulted to. 

My apologies for this and no further comments regarding this issue will be made. 

~Ibuprophen

CyberWeather, 2018-08-05:
@ibuprophen1 @pbatard Thank you for the help though. Really appreciate it!

Jubliano, 2018-08-13:
I have the same issue and now when I put the usb stick into any windows 7 devices I get the BSOD and a win auto reboot. If I try to boot from the USB it flashes op something (too quick to read) and reboots!! Probably I have given up on Free NAS  (I'll set up the PC  on a Lan and use team viewer for remote access) BUT how the hell do I recover my USB stick since every time I insert it in win 7 device I get the BSOD??

Jubliano, 2018-08-13:
ok so I have now partially answered my own question. USB recovered as follows: loaded old copy of Zorin (linux derivative) from another mem stick and the original stick was then readable (which tends to suggest it is a win 7 issue) 
and I could format it and recover the drive. Out of interest could someone tell me what is wrong (or what I did wrong) with the way the memory stick is configured (see attachment). Maybe that's why it would not boot from it anyway 
[image]

pbatard, 2018-08-13:
> could someone tell me what is wrong with the way the memory stick is configured

Sorry if that sounds a bit harsh, but you shouldn't piggyback on an issue opened with regards to a specific application (Rufus) to ask support for software that is unrelated to that application (FreeNAS), __especially__ if that issue is closed.

If you have a query on how to configure a bootable thumb drive for FreeNAS, then please ask that question on the FreeNAS forums, as it will be A LOT more helpful __for others who may have the same question__ if you do so there (as very few FreeNAS people are expected to check closed issues on unrelated software to find out if their bootable drive is configured properly).

Thanks.