pbatard/rufus #1191: Rufus stuck on  Clearing MBR/PBR/GPT structures

Kemsing531, 2018-08-11:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x ] I also tried one or more of the following:
  - [ ] Using a different USB drive. - no others large enough
  - [x ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer. - no other computer owned
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When trying to burn an ubuntu 18.04 iso file to my usb stick, rufus gets stuck on "Clearing MBR/PBR/GPT structures..." and never leaves off it
Log
---
```
<FULL LOG>
```Rufus version: 3.1.1320
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0809
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Generic Mass Storage USB Device' (1908:9999)
1 device found
Disk type: Removable, Disk size: 34GB, Sector size: 512 bytes
Cylinders: 4171, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT32 (0x00)
  Size: 32.0 GB (34309406720 bytes)
  Start Sector: 0, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 18.04.1 LTS amd64'
  Size: 1.8 GB (Projected)
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-18.04.1-desktop-amd64.iso (1.8 GB)
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 18.04.1 LTS amd64'
  Size: 1.8 GB (Projected)
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-18.04.1-desktop-amd64.iso (1.8 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20171017/' for Syslinux installation

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE3 for exclusive write access
Requesting lock...
Will use 'E:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Cancelling
Could not reset partitions
Re-mounted volume as 'E:' after error

Found USB 2.0 device 'Generic Mass Storage USB Device' (1908:9999)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 34GB, Sector size: 512 bytes
Cylinders: 4171, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 32.0 GB (34309406720 bytes)
  Start Sector: 0, Boot: No
Closed Windows format prompt
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20171017/' for Syslinux installation

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE3 for exclusive write access
Requesting lock...
Will use 'E:' as volume mountpoint
Requesting lock...
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors

pbatard, 2018-08-11:
If it doesn't error out, it means that your drive access might be very slow for some reason. have you simply tried waiting? If there is an issue, Windows should time out or report an error on the write, and the fact that you are seeing the `Erasing 128 sectors` message means that we are in the part of the code where all we are doing is tell Windows to write to those sectors.

I have yet to see a system where a write operation would just freeze without resulting in a timeout or an error, and, at any rate, even if that were the case, this isn't something that could really be fixed in Rufus, as all we are doing at this stage is tell Windows to write zeroed data on a bunch of sectors, and relying on the Windows APIs to do so. If that doesn't work, then it looks like a Windows problem.

So I have a few questions:
- What happens if you try to clear the drive using `diskpart`? (if you don't know how to do that, you should be able to find tutorials through google)
- What happens if you use a different flash drive?
- Did you __really__ run a bad blocks check?? I find it very curious that the bad block check would not produce the same issue whereas it's basically going to be doing the same thing. Can I see a log with the actual report from the bad blocks check?
- What happens if you try to write the Ubuntu image in DD mode? You should have gotten a  prompt asking you whether you wanted to write the image in ISO or DD mode. Did you try both options?
- What happens if you try to zero the drive with the <kbd>Alt</kbd>-<kbd>Z</kbd> shortcut?

Kemsing531, 2018-08-11:
ive done the diskpart, its currently reformatting right now
bad blocks check was on 4 passes
i dont have another flash drive
haven't tried dd, should i?
haven't zero'd, not sure how

pbatard, 2018-08-11:
> haven't tried dd, should i?

It's what I advised you to do, so yes.

> haven't zero'd, not sure how

I told you how. Press <kbd>Alt</kbd>-<kbd>Z</kbd> in Rufus.

pbatard, 2018-08-13:
Any update on this? Did DD mode work? How about zeroing the drive?

pbatard, 2018-08-14:
No update, so I will consider that this issue can be closed.

closed 2018-08-14 by pbatard

bertvangelder, 2019-01-23:
FYI
I had the same problem, and when trying to zero the drive, it was also stuck on 0.0%
Switched to a different USB port and this did the trick. I guess this USB-port isn't working properly anymore.
Zero-ing right now, hopefully the rest will work too. I need Win10 for my new PC, and since USB-ports are failing it's way overdue ;)