pbatard/rufus #1193: Rufus Error writing Reflection

DrAvengers, 2018-08-13:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I apologize for my bad english first

I encounter errors when printing a disc after updates. When I select the legacy partition format, the disk format is automatically changed to NTFS. If you are doing it manually, you get the above error, so I keep using the old version and need to fix it .

Log
---
```
<FULL LOG>
```

pbatard, 2018-08-13:
Please provide a log. I have no idea what `Error writing reflection` means since there is no English message like this.

Also, if Rufus uses NTFS, then I can guarantee that your flash drive can boot with NTFS.
Have you actually tried booting the NTFS drive. Do __NOT__ believe the reports you see on the internet telling you that FAT32 is mandatory. You __can__ use NTFS, even with UEFI, and Rufus always selects the file system that will work best.

pbatard, 2018-08-15:
Yes, we changed the default because a lot more people use UEFI-based systems these days than BIOS-based ones, so, because there is no way for Rufus to "guess" whether your target system is BIOS or UEFI, the new defaults will favour UEFI instead of BIOS.

It will always be the user's job to find out if their target is BIOS or UEFI and select the right option in Rufus.

At any rate, I'm happy to hear that you solved your problem and I will therefore close this issue.

closed 2018-08-15 by pbatard