pbatard/rufus #1195: ISO extraction failure

george2781, 2018-08-14:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2018-08-14:
> - [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.

If you really did that, then there should be a log. Please provide the full log, or I will have to close this issue.

george2781, 2018-08-14:
trying yet can't get log

pbatard, 2018-08-14:
Do you have an issue accessing the log in Rufus? Or do you have an issue copy/pasting the log into the github issue tracker?

As explained in the steps, you just need to click the last small button before _START_ to get the log window to display in Rufus, and once you have it you can click _Save_ to save the log to a file.

Then, if github doesn't let you copy/paste the content of the file, you can always attach it to the issue by dragging and dropping the file.

george2781, 2018-08-14:
accessing the log

pbatard, 2018-08-14:
[image]

george2781, 2018-08-14:
ok got it thank you

george2781, 2018-08-14:
might take a minute

pbatard, 2018-08-14:
No worries. Take your time.

george2781, 2018-08-14:
ok got it

pbatard, 2018-08-14:
So the issue is that Rufus appears to freeze after the `Requesting lock...` message?

I thought the issue was a file extraction failure (which would produce a different message in the log).

Are you sure you copy/pasted the whole log?

george2781, 2018-08-14:
yes

george2781, 2018-08-14:
[image]

pbatard, 2018-08-14:
Also, looking further at the log, I think your issue is pretty obvious:

```
ERROR: Detected that file on disk has been truncated by 1.2 GB!
```

Means that your download is __corrupt__.

Also, the SHA-256 you got does not match with the official SHA-256 for Ubuntu 18.04.1. You got `1ec698b4a55c6d8e74f171a01b5e5d6530276f45f5ef96bd8857445bba0cd688`, whereas the correct one is `5748706937539418ee5707bd538c4f5eabae485d17aa49fb13ce2c9b70532433`.

Corrupt download. You'll need to sort that out on your own, as it has nothing to do with Rufus.

Next time, please pay a bit more attention to the checklist, especially the part that says, about the checksum:

> I confirmed, by performing an internet search, that these values match the ones from the official image.

pbatard, 2018-08-14:
For what is worth, you can find the proper checksums [here](http://releases.ubuntu.com/18.04/SHA256SUMS)

closed 2018-08-14 by pbatard