pbatard/rufus #1204: Feature: Show progress on individual file copying

vanarebane, 2018-09-05:
As we can see in issue #746 users will assume that Rufus is stalled when copying large files.
So here is a feature request, to show progress by copied file size, rather than file number.

Thank you!

pbatard, 2018-09-05:
There __already__ is a progress bar &rarr; Rejected.

closed 2018-09-05 by pbatard

vanarebane, 2018-09-05:
I did not mean to add another progress bar. I meant to show better information in the progress bar. Right now, as I understand, the progress bar moves when a file is copied. In the case of windows server 2016 iso, where there is one 5.6GB file amongst smaller ones, the progress bar jumps forward (from ~20% to ~80%) when the 5.6GB file comes up. The progress bar stays in one place for 10 minutes (showing no progress). Me as a user, I figure that something is wrong and start to investigate.

So what I meant was for the progress bar to show how many % of the file MB/GB are copied and not to just jump forward when a large file comes up and stay at that position for long time.

pbatard, 2018-09-05:
> In the case of windows server 2016 iso, where there is one 5.6GB file amongst smaller ones, the progress bar jumps forward (from ~20% to ~80%) when the 5.6GB file comes up.

I have not observed that.

>  The progress bar stays in one place for 10 minutes (showing no progress). 

I have not observed that at all either.

Therefore, as I indicated, the issue is entirely on your side. If you are using slow hardware, that doesn't report progress, then there's nothing that can be done.

We already report progress according to the best information we can have, which is Windows indication of how many bytes were copied. There's no additional extra "magical" indicator that we can use here, so if you saw no progress, then it means that there's __REALLY__ nothing more we can do, because there is no additional progress indicator that we can use here. I am not going to start pretending that bytes have been copied, just to move the progress bar, when there is __NOTHING__ that Windows can report that indicates so. Especially when Windows flushes its cache(s), you don't get to find out how many bytes were copied or how long that will take. The only data you get is how many bytes the system sees were copied, and that is what we already use.

But then again, with decent hardware, I have no observed the "freezout" while copying you report, so I will continue to suggest that you use better hardware, especially as, if the issue is with your flash drive controller freezing as it reorganizes bad blocks or for any other reason, there is __absolutely__ no indication of progress you can ever get on that operation.

vanarebane, 2018-09-05:
I'm clad that you are using the best there is. You are right, there is no stalling in Rufus, but what I saw was a progress bar not moving for a solid 10 minutes and then it moved again. Call me crazy for believing that an average user would assume that nothing is moving (even while in the background, everything was progressing nicely) 

If this problem is with Windows API's, then it's not your fault. I will record a video to demonstrate the problem and maybe you can understand a bit better what is going on.

pbatard, 2018-09-05:
> Call me crazy for believing that if an average user sees nothing moving for 10 minutes (even while in the background, everything was progressing nicely), they would not assume the coping is stalled.

You are misrepresenting my argument, which is a very deceitful thing to do. That is not at all what I did. I have never disputed that you saw a stall. All I said is that I have not observed one with decent hardware, and drew the only __logical__ conclusion one can draw from it, which is that the issue has probably to do with you using slow hardware.

I do get that you are frustrated, but using a straw man, as you are doing, is not going to get you anywhere. On the contrary, if this is how the discussion is going to it makes me a lot more inclined to cut it short.

> If this problem is with Windows API's, then it's not your fault.

It is with the Windows API. If you know something I don't about being able to monitor cache flushing (without disabling cache altogether, which would kill write performance for most people, and without spying on mass storage commands sent to the USB bus, which would be a complete overkill just to second guess what Windows is doing), then I'm all ears. Otherwise, you will just have to accept that Rufus is already doing the best it can in terms of reporting progress and that any stall you experience comes from a Windows limitation with regards to how it doesn't always match the apparent write progress with the actual one, rather than Rufus.