pbatard/rufus #1206: Rufus 3.2 no longer recognizes raspian image as valid

Jeansen, 2018-09-12:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus 3.2 does not recognize raspian image. Seel log. Version 3.1 works, though!

Steps to reproduce: 
- Download RASPBIAN STRETCH WITH DESKTOP https://downloads.raspberrypi.org/raspbian_latest
- Check SHA256 to match https://www.raspberrypi.org/downloads/raspbian/
- Extract the image
- Start rufus 3.2 and select the image. Check the log.
- Start rufus 3.1 and select the image. Check the log.

Log v3.2
---
```
Rufus version: 3.2.1397
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0407
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found USB 3.0 device 'TS-RDF5 SD  Transcend USB Device' (8564:4000)
Using autorun.inf label for drive E: '32 GB'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3880, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001E36EC
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.7 GB (31913934848 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  'C:\Users\Marcel\Desktop\raspbian_latest\2018-06-27-raspbian-stretch.img' doesn't look like an ISO image
Disk image analysis:
  Image has a Zeroed Master Boot Record
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3880, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001E36EC
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.7 GB (31913934848 bytes)
  Start Sector: 2048, Boot: No
Failed to scan image
```

Log v3.1
---
```
Rufus version: 3.1.1320
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0407
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found USB 3.0 device 'TS-RDF5 SD  Transcend USB Device' (8564:4000)
Using autorun.inf label for drive E: '32 GB'
1 device found
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3880, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001E36EC
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.7 GB (31913934848 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  'C:\Users\Marcel\Desktop\raspbian_latest\2018-06-27-raspbian-stretch.img' doesn't look like an ISO image
Disk image analysis:
  Image has a Zeroed Master Boot Record
  Image is a bootable disk image
Using image: 2018-06-27-raspbian-stretch.img (4.5 GB)
```

pbatard, 2018-09-12:
Thanks for the report. I can confirm the bug.

Just so you know, if you use the `.zip` of Raspbian instead of the uncompressed `.img` (which Rufus supports) you won't have the issue.

After a quick analysis, I have a good idea why that happens, and I will probably release a bugfix 3.3 as a result, since this is a pretty major issue.

Jeansen, 2018-09-12:
I am glad I could help and thank you for the tip regarding `.zip`. I did not know that one :smile:

pbatard, 2018-09-13:
The issue has now been fixed in the code. I am planning to release a bugfix version of Rufus 3.3 that includes this fix sometime tomorrow or early next week.

closed 2018-09-13 by pbatard

pbatard, 2018-09-14:
For the record, there's a BETA of Rufus 3.3 available [here](https://rufus.ie/downloads/), that includes the fix for this issue, if you want to check it out.

commit 2d99601a (2018-09-13): [core] fix a regression when scanning uncompressed bootable DD images

* Closes #1206