pbatard/rufus #1207: Unable write OpenBSD USB image

dmitry-sensei, 2018-09-14:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Aplication version 3.1and prior  allowed to write images of OpenBSD for a USB (install64.fs). It is non-iso image but disk image (for dd if... of=/dev/sda...)

Log
---
```
<FULL LOG>
Rufus version: 3.2.1397
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (8564:1000)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1899, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x71FFADA2
Drive has an unknown Master Boot Record
Partition 1:
  Type: EFI FAT (0xef)
  Size: 480 KB (491520 bytes)
  Start Sector: 64, Boot: No
Partition 4:
  Type: OpenBSD (0xa6)
  Size: 359.5 MB (376963072 bytes)
  Start Sector: 1024, Boot: Yes
Scanning image...
ISO analysis:
  'D:\dmitriy.o\Downloads\Images\install64.fs' doesn't look like an ISO image
Disk image analysis:
  Image has an unknown Master Boot Record
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1899, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x71FFADA2
Drive has an unknown Master Boot Record
Partition 1:
  Type: EFI FAT (0xef)
  Size: 480 KB (491520 bytes)
  Start Sector: 64, Boot: No
Partition 4:
  Type: OpenBSD (0xa6)
  Size: 359.5 MB (376963072 bytes)
  Start Sector: 1024, Boot: Yes
Failed to scan image

ItielMaN, 2018-09-14:
Already fixed in #1206 if I'm not mistaken.

pbatard, 2018-09-14:
Yes, this is #1206. Because this is a rather unfortunate bug, I am planning to release a BUGFIX release of Rufus 3.3 early next week for that.

In the meantime, you can use Rufus 3.3 available [here](https://rufus.ie/downloads/).

closed 2018-09-14 by pbatard