pbatard/rufus #1213: UEFI:NTFS - Unable to start driver: [1] or Could not start NTFS partition service: [14]

zanechua, 2018-09-21:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Motherboard: S2600CP4
Bios Revision: R02.06.E006
Rufus version: 3.3

Here's the log from Rufus:

Here are the errors I am getting on boot:
[image]

[image]

EDIT: 

So it seems like the last version of Rufus v2 works.

Rufus version: 2.18

Here's the log from Rufus:
https://gist.github.com/zanechua/acf8c67c7026814796f7da6414dd6e3e

[image]

pbatard, 2018-09-21:
Thanks for opening a new issue.

As I mentioned in the previous one, I suspect that the problem comes from the EfiFs NTFS driver having switched from using the `IMAGE_SUBSYSTEM_EFI_RUNTIME_DRIVER` [PE type](https://docs.microsoft.com/en-us/windows/desktop/api/winnt/ns-winnt-_image_optional_header) to `IMAGE_SUBSYSTEM_EFI_BOOT_SERVICE_DRIVER`, which I suspect some UEFI Firmwares might not like. The problem however is that I had to change that attribute because it was creating issues with other UEFI firmwares, and `IMAGE_SUBSYSTEM_EFI_BOOT_SERVICE_DRIVER` is the proper type to use for what UEFI:NTFS is doing.

You are mentioning that you went through a BIOS update, so I wonder if that update is preventing EFI executables with `IMAGE_SUBSYSTEM_EFI_BOOT_SERVICE_DRIVER` from running.

Here's what you can do to validate that this is indeed the issue, using the USB you created with Rufus 3.3:
- Download CFF Explorer from [here](https://ntcore.com/?page_id=388). If you don't know which version you should pick, I'd recommend the _CFF Explorer (x86 Version, stand-alone, Zip Archive)_ as this is smaller and it doesn't need to be installed.
- Plug in the Flash Drive created by Rufus. Since you're running on Windows 10, you should be able to access both the partitions Rufus created on that drive, including the UEFI:NTFS. However, Windows may not mount it automatically, and Disk Management seems to have trouble displaying small partition to do the following with the GUI, in which case you will have to assign a letter manually using `diskpart`. In case you cannot see the `UEFI_NTFS` partition, you should:
  - Open an elevated command prompt
  - Run `diskpart`
  - Run `list vol`. This will list all the volumes available, and you should see something like `Volume 7         UEFI_NTFS    FAT    Removable    512 KB  Healthy` in your list. Take a note of the volume number for the UEFI_NTFS partition. 
  - Run `sel vol #` where `#` is the number above (e.g. `sel vol 7`)
  - Run `assign letter=X` where `X` is any drive letter that is not already in use. After that, you should be able to access the UEFI:NTFS partition from Windows Explorer. A screenshot of the procedure is also provided below:

    [image]

- With both CFF Explorer and the UEFI:NTFS partiton available, launch CFF Explorer
- Open `EFI\Rufus\ntfs_x64.efi` on the UEFI:NTFS partition
- On the left pane navigate to _Nt Headers_ &rarr; _Optional Header_. This should allow you to view the _Subsystem_ property and see that it is set to _EFI Boot Driver_:

  [image]
- Click on _EFI Boot Driver_ and change it to _EFI Runtime Driver_ and then click the _Save_ button (select _Yes_ when it asks you to overwrite the file)
- Try to boot the USB again and see if that works better.

Of course, this is just one possibility, and I may be off mark (especially it's possible that I had already changed the subsystem type before 2.18, in which case the above probably won't help).

The other thing that I'd like you to try, to confirm that this has to do with the EFI driver, is to pick the  `EFI\Rufus\ntfs_x64.efi` from the 2.18 drive and copy it to the 3.3 drive (replacing the existing one) and see how that work. Alternatively, you can try each of the `ntfs_x64.efi` from https://efi.akeo.ie/downloads/, starting with EfiFs 1.0 (and then selecting the `x64` repository) to see which one works and which one doesn't, as this may help me pinpoint what the issue might be.

zanechua, 2018-09-22:
Thanks for the very very detailed guide on how to debug this specifically.

The ntfs_x64.efi file from the 2.18 drive already has the Subsystem header set to EFI Boot Driver

[image]

I changed the Subsystem header to EFI Runtime Driver like you said but got a different error this time.

[image]

However copying the file from 2.18 to the drive created by 3.3 allows the drive to boot perfectly fine.

pbatard, 2018-09-22:
Thanks for the test. I probably should have checked the driver from 2.18 before I asked you to run that test.

As I mentioned earlier, can I ask you to please try all of the drivers `ntfs_x64.efi` from https://efi.akeo.ie/downloads/, starting from `efifs-1.0/` and let me know the last version that works with your 3.3 flash drive?

zanechua, 2018-09-23:
Oh. I misssed that part sorry. Will get to it.

pbatard, 2018-09-25:
Thanks for the update, but next time please don't edit a comment to add new data, as I didn't see your tests with the various version until now! It's much better to create a new entry, as I do get a notification.

There were a lot of changes between 1.1 and 1.2, and since you are the only person with a machine that can replicate the problem, this is probably going to take a lot of back and forth to troubleshoot, so it may take a while.

I guess the first thing I'm going to test, since 1.2 changed the compiler being used, it to try with a different compiler. I can not revert to the old one (Clang/C2) since support for it is being dropped by Microsoft in the latest version of Visual Studio, which is part of the reason why I switched compiler for the binary in the first place. Since version 1.2, the NTFS driver binary is compiled with the EDK2, so a good test will be to try non EDK2 versions.

If you look in https://rufus.ie/testing/ you'll see that I uploaded 2 drivers: `ntfs_x64_gcc.efi` and `ntfs_x64_msvc.efi`. These are the latest versions of the NTFS driver, but compiled with GCC and MSVC respectively. Can I please ask you to test each one (by removing the `_gcc` or `_msvc` part and replacing your existing `ntfs_x64.efi`) and see how it goes?

zanechua, 2018-09-25:
Sure.

Can I verify that the hash of the ntfs_x64.efi file in the v3.3 of rufus is 872FC3150C9DC39470CF0E03732642789CF8F82F? and it's 47KB?

Both testing drivers work totally fine and boot up as they should. Only the one included with rufus v3.3 doesn't.

!(success_v13)[https://i.imgur.com/fheglqp.jpg]

pbatard, 2018-09-25:
Many thanks for the new tests!

Yes, the sha1sum of `ntfs_x64.efi` from Rufus 3.3 is `872fc3150c9dc39470cf0e03732642789cf8f82f`. And it's 46.3 KB in size. Basically, you can find the file included in Rufus if you extract it from the current `uefi-ntfs.img` we have in our source tree at https://github.com/pbatard/rufus/tree/master/res/uefi (using something like 7-zip for instance).

It's quite interesting that the test MSVC version works, whereas the original one doesn't, as the EDK2 version used by Rufus uses MSVC for compilation under the scenes.

So, just to make absolutely sure that this is not an EDK2 issue that has since been fixed, can I please ask you to try one more test? I have just recompiled a new EDK2 driver, which you can find as `ntfs_x64_edk2.efi` under https://rufus.ie/testing/. Can you run the same test with that one as you did with the others, and confirm that it fails?

If it does, I guess I'll have to figure out the differences between the way the EDK2 invokes MSVC and the way the regular Visual Studio solution does (besides the use of gnu-efi, which I would assert should be the main difference), so that I can try to produce a driver that works.

Else, I guess I'm going to have to switch to compiling all the drivers with Visual Studio + gnu-efi instead of EDK2, which is fine, but which will increase the size of the binaries and therefore the size of Rufus (the reason I'm using EDK2 + MSVC is because it produces the smallest binaries). And it still won't tell me exactly what it is between your platform and these drivers that is causing this weird incompatibility...

zanechua, 2018-09-25:
I can confirm that the EDK2 driver fails.

If you need more information let me know and I'll do my best to provide it.

pbatard, 2018-09-25:
Thanks.

I think right now I need to process all this good data you provided, and come up with some new things to try, to zero the issue further (which probably won't happen be before next week at this stage).

At any rate, your testing and your patience with this issue are much appreciated!

Mattiwatti, 2018-09-29:
I can't reproduce this myself (of course...), but EDK2 does make quite a few drastic changes to the standard MSVC build environment in the interest of saving space. I'm going to go ahead and assume EDK2 itself is not at fault (meaning the headers as opposed to the gnu-efi ones) as I've used both for years with no issues. But comparing `ntfs_x64_edk2.efi` vs`ntfs_x64_msvc.efi`/`ntfs_x64_gcc.efi`, there are a number of differences the DXE core loader could be stumbling over in the EDK2 PE files compared to the MSVC/GCC ones:
- File and section alignment. These are set to 0x20 by default by EDK2, which is very very aggressive. Try the default MSVC `/ALIGN:0x1000 /FILEALIGN:0x200` first and see if that helps. This is IMO the most likely culprit.
- No image base. Probably not the cause since the image should be relocated, but link with `/BASE:0x180000000` just in case.
- Subsystem version not set (should be 1.0): link with `/SUBSYSTEM:EFI_BOOT_SERVICE_DRIVER,1.0` to correct this.
- Checksum not set: it could be that OP's BIOS is being pedantic about this (but if so, it seems strange to me that it's not complaining about the file being unsigned). You can link with `/RELEASE` to set a checksum, though this will most likely be wrong because EDK2 runs some post-build steps that alter the image. Use something like CFF Explorer to fix this as the very last step.
- `SizeOfStackReserve`, `SizeOfStackCommit`, `SizeOfHeapReserve` and `SizeOfHeapCommit` have all been zeroed. I have no idea if this matters for OP's BIOS (odds are it doesn't) but I would just set these to `0x100000`, `0x1000`, `0x100000` and `0x1000` respectively with a hex editor to be on the safe side.
- The debug directory contents have been wiped by EDK, but the PE debug directory still exists and points to empty data. Zero 8 bytes at offset `0x168` to get rid of this entry.
- The exception directory has been removed. I highly doubt this matters as there is no SEH in EFI, but on the other hand, the MS boot loaders do keep this directory around and they probably thought about this. I dunno.

To get most of the above in order you can put the following in the `.inf` file (obviously applies only to that package - for 'affects everything' scope edit `tools_def.txt` instead):
```inf
[BuildOptions]
  MSFT:*_*_*_DLINK_FLAGS = /ALIGN:0x1000 /FILEALIGN:0x200 /BASE:0x180000000 /SECTION:.pdata,!D /MERGE:.rdata=.text /SUBSYSTEM:EFI_BOOT_SERVICE_DRIVER,1.0 /RELEASE
  MSFT:*_*_X64_GENFW_FLAGS = --keepexceptiontable --keepzeropending --keepoptionalheader
```

pbatard, 2018-09-30:
@Mattiwatti, many thanks for the extended analysis. This should indeed be exceedingly helpful, and I'll try to produce and upload drivers that apply your suggestions, to see if we can validate your hypotheses.

jjboy32, 2018-10-02:
文件名 cn_windows_10_business_edition_version_1803_updated_aug_2018_x64_dvd_57e5b984.iso SHA1 533a5c7a732aefc7025f2128ffe0a4f0e5b0ed46 文件大小 4.86GB 发布时间 2018-08-22 ed2k://|file|cn_windows_10_business_edition_version_1803_updated_aug_2018_x64_dvd_57e5b984.iso|5221404672|DD0094E00A4A1D7FB36D2FBBB82DCAED|/

This is the latest version of win10 x64 in Chinese, and the English version is the same. Rufus is very good to engrave it into the U disk, which is very good in dealing with the problem of ntfs+uefi starting the U disk. When win10 is installed, GPT disk will be installed. If the NTFS format is not directly engraved into the UEFI disk, the win10 installation will not be installed to the GPT disk after it is installed. Thank you.

jjboy32, 2018-10-02:
In addition, there is a TF reader [Chuanyu USB3.0 TF reader], which is read normally under win10 x64 1803. But Rufus can't read the device. The information is as follows:

0 devices found
Found USB 3.0 device 'SD Card Reader USB Device' (067B:2731)
Device eliminated because it was detected as a Hard Drive (score 7 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found

pbatard, 2018-10-02:
@jjboy32, you are piggybacking on an __unrelated__ issue, therefore I will have to delete your comments. Also I'm afraid the nature of your problem is not very clear. If you want to report something, please open a new issue and make sure you provide a full log as well as a description of the actual error.

You may also want to read the "Limitations" paragraph points regarding the use of UEFI:NTFS as described [here](https://github.com/pbatard/uefi-ntfs).

I will delete your comments as well as this one in a few hours.

pbatard, 2018-10-04:
Finally got around to compile a NTFS driver that includes @Mattiwatti's build options suggestion. So let's find out if it's really one of these compilation settings that is really causing the issue or something else.

The driver is called `ntfs_x64_edk2_TEST1.efi` and can be found at the [usual place](https://rufus.ie/testing/).

@zanechua, if you can conduct a test with this driver, as you did with the other ones, I would realy appreciate it.

Also, for the record, the linker issued the following warning, which means that the `/MERGE:.rdata=.text` was most likely ignored. But since the other parameters should have been applied that doesn't worry me too much:
```
LINK : warning LNK4253: section '.rdata' not merged into '.text';  already merged into '.data'
```

Mattiwatti, 2018-10-05:
That warning is not an issue, it just means some other EDK2 build setting already applied `/MERGE:.rdata=.data` instead. Both options will merge the .rdata section into another one, saving space, but neither of them is either harmful or useful as far as OP's issue goes.

The only difference is that `/MERGE:.rdata=.text` will make .rdata contents executable, which is usually not a problem. `/MERGE:.rdata=.data` will make .rdata contents writable, which can cause issues when debugging because .rdata is supposed to be for read-only data (hence the name) and any bug where you accidentally write to `const` data won't be detected. If your debug build configuration does not merge .rdata into .data then this is not an issue.

zanechua, 2018-10-05:
Tested ntfs_x64_edk2_TEST1.efi and I still get the same error.

pbatard, 2018-10-05:
Thanks for the test, it was definitely worth a try. My guess would still have to be that EDK2 has deprecated APIs that gnu-efi hasn't, therefore I will replace all the x86 drivers with their gnu-efi version in the next release of Rufus. For the sake of not increasing the size of the Rufus executable too dramatically, I think I'll keep the ARM/EDK2 versions though, as I would surmise that any ARM UEFI firmware should be modern enough to work with drivers compiled with a recent EDK2.

pbatard, 2018-10-19:
@devedse, can you please indicate if you're getting errorcode 1 or 14, or some other.
Even if you are getting the same error, it always helps to copy the exact message you are getting, just to confirm that you are indeed seeing the same error.

devedse, 2018-10-19:
Didn't manage to make a screenshot as I went back to 2.18 and that worked. I'll see if I can create a new usb stick with the 3.3 version.

devedse, 2018-10-19:
I tried 2 diffent UEFI files on the disk, 1:
[image]
[image]

And 2:
[image]

(Just a black screen)
[image]

pbatard, 2018-10-20:
Well, for 2 that's a given. `ntfs_x64.exi` is a __driver__, not a bootloader. You can't boot that.

As to the first test, you are getting a completely different error from the ones mentioned above. Interesting...

> Also one thing I don't quite understand is that I can select fat32 in Rufus 2.18 but not in 3.3.

Well, UEFI:NTFS is of course __only__ valid for NTFS, so obviously it makes no sense to have it with FAT. I sure hope you didn't use FAT32 for your 2.18 test, because then it makes your success there irrelevant.

pbatard, 2018-10-20:
Also, why does the boot option name in your first screen says `WinUsb2`?
I'm sorry but your tests look super weird.

Care to post a full log of how you created the USB with Rufus 3.3?

devedse, 2018-10-20:
I did use NTFS for both tests. The WinUsb2 is just a name that you can manually enter.

Full process would be:
1. Download img
2. Download rufus
3. Start rufus
4. Load image in rufus without changing any settings.
5. Open bios of dell laptop and add uefi boot option.
6. Browse to uefi file as mentioned above on the usb drive

Same for 2.18 but there it did work.

pbatard, 2018-10-20:
Still, I would __really__ like to see the full log of you created the USB with Rufus 3.3.
Remember, the more data I have, the more I am able to help __you__.

devedse, 2018-10-20:
```
Rufus version: 3.3.1400
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'USB Device' (058F:6387)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1950, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {4336111D-E9A7-45D7-A847-8ACB4B098C78}
Max parts: 128, Start Offset: 17408, Usable = 16043178496 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {34813333-0910-43E2-86E5-F11B37B53464}
  Size: 14.9 GB (16041070080 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {C720D919-0AD9-401C-B039-7EC08C65C37F}
  Size: 512 KB (524288 bytes)
  Start Sector: 31332263, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CPBA_X64FRE_EN-US_DV5'
  Size: 4.9 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: en_windows_10_business_edition_version_1803_updated_sep_2018_x64_dvd_37051f54 (1).iso (4.9 GB)
Checking for Rufus updates...
Checking release channel...
Updates: Unable to access version data

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Found volume GUID \\?\Volume{cc0f16d3-d461-11e8-9cb2-34f39a973ded}\
\\?\Volume{cc0f16d3-d461-11e8-9cb2-34f39a973ded}\ was already mounted as D:\
[... 1007 lines cut ...]
   3916276 total allocation units on disk.
   2633271 allocation units available on disk.
NTFS Fixup completed.

Found USB 2.0 device 'USB Device' (058F:6387)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1950, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {EBC4A794-38DF-469D-8D80-BA4F8CF2388C}
Max parts: 128, Start Offset: 17408, Usable = 16043178496 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {6638616F-F479-4799-BB6B-16648722C431}
  Size: 14.9 GB (16041070080 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {021EBE54-FA96-4E80-99C3-AD11F531D068}
  Size: 512 KB (524288 bytes)
  Start Sector: 31332263, Attributes: 0x0000000000000000

```

pbatard, 2018-10-20:
Appreciate that - thanks!

devedse, 2018-10-21:
Let me know if you need anything else

pbatard, 2018-10-24:
Okay, I have now put a TEST version of Rufus 3.4, that contains the update gnu-efi based NTFS drivers, that I expect to fix @zanechua's original issue.

Can you please download `rufus-3.4_TEST1.exe` from [here](https://rufus-web.akeo.ie/testing/rufus-3.4_TEST1.exe) and report?

zanechua, 2018-10-26:
@pbatard sorry. give me a little bit. been busy with life.

devedse, 2018-10-26:
> Well, UEFI:NTFS is of course **only** valid for NTFS, so obviously it makes no sense to have it with FAT. I sure hope you didn't use FAT32 for your 2.18 test, because then it makes your success [...]

Btw, does this mean that FAT32 functionality was removed from Rufus 3.3? As I can't find the option anywhere to select it.

pbatard, 2018-10-26:
> does this mean that FAT32 functionality was removed from Rufus 3.3?

Not at all. But obviously, Rufus removes the ability to create a FAT32 drive __IF__ it detects that there is a >4GB file on your image, since it would only produce an error if you tried it (because you won't be able to copy a >4GB file onto a FAT32 file system without truncating it, which means your drive won't work).

devedse, 2018-10-26:
> Not at all. But obviously, Rufus removes the ability to create a FAT32 drive **IF** it detects that there is a >4GB file on your image, since it would only produce an error if you tried it (because [...]

For 2.18 and this file I could still select FAT32. But I expect that it would error out later in the process then?

pbatard, 2018-10-26:
You're most likely not using the same partition scheme and target in 2.18 as you do in 3.x. Please bear in mind that the __defaults__ have changed between 2.18 and 3.0.

Anyway, if you think there is a problem with this, __PLEASE__ open a new issue, as this is completely unrelated to the matter at hand, and this thread is long enough as it is without hijacking it for a matter that is not directly relevant.

devedse, 2018-10-26:
@pbatard, I thought opening a completely new thread would also cause some overhead for a simple question.

Back on topic though, I just tested your test version and atleast in my case it works and boots to the Windows Setup.

pbatard, 2018-10-26:
> I just tested your test version and atleast in my case it works and boots to the Windows Setup.

Great. Thanks for the test.

jjboy32, 2018-10-29:
ed2k://|file|cn_windows_10_business_edition_version_1803_updated_aug_2018_x64_dvd_57e5b984.iso|5221404672|DD0094E00A4A1D7FB36D2FBBB82DCAED|/

I met WIM ISO in file win10 is greater than 4G problem. You have to burn to the U disk, other burning software burned after the U disk is NTFS system, then the U disk boot installation system, if your hard disk is GPT format, you can not install, because it is EFI boot, EFI does not recognize ntfs. If your hard disk is in FAT32 format, it can be installed, but it can only be started by RBM. We need uefi+gpt to boot. At this point, Rufus 3.3 solves the problem by carving the U disk into two partitions, starting efi, and installing gtp. As for the FAT32 format, is not it?

pbatard, 2018-10-29:
@jjboy - Please don't add unrelated comments to an existing thread. This thread is for people who experience the same issue as the one reported by OP. I will deleted your comment (and this one) in one hour.

jjboy32, 2018-10-29:
ed2k://|file|cn_windows_10_business_edition_version_1803_updated_aug_2018_x64_dvd_57e5b984.iso|5221404672|DD0094E00A4A1D7FB36D2FBBB82DCAED|/  I encountered the problem of WIM file larger than 4G in win10 ISO. You have to burn to the U disk, other burning software burned after the U disk is NTFS system, then the U disk boot installation system, if your hard disk is GPT format, you can not install, because it is EFI boot, EFI does not recognize ntfs. If your hard disk is in FAT32 format, it can be installed, but it can only be started by RBM. We need uefi+gpt to boot. At this point, Rufus 3.3 solves the problem by carving the U disk into two partitions, starting efi, and installing gtp. As for the fat 32 format is not, we do not know, it can not be used for a single file greater than 4GB.

pbatard, 2018-10-29:
Awesome! I will close this issue then, as the next release of Rufus should resolve the issue. Many thanks for your testing and your patience.

pbatard, 2020-04-15:
Unlocking this thread to give a heads up to the people who were affected that I have switched the UEFI:NTFS file system driver to the EDK2 version in [Rufus 3.10 BETA](https://github.com/pbatard/rufus/releases/tag/v3.10_BETA).

The reason is that I am hoping that the recent improvements and bugfixes that were applied to the file system drivers may fix the issue reported above, and using a BETA is a good way to confirm whether that is the case or not, so that, if the problems still persist, I can switch back to the gnu-efi version of the drivers in time for the release.

If you were affected by this, and if you still have an opportunity to test, may I ask you to please try to create an UEFI:NTFS bootable media with Rufus 3.10 BETA, and report if you see a regression? Thank you.

zanechua, 2020-04-15:
I would love to test this for you but unfortunately the board I was using had a cap blow up and I haven't tried booting it since. If I can replace the cap I might be able to try but it probably won't be anytime soon. Unless @devedse can do it.

pbatard, 2020-04-15:
No worries. I still appreciate that you took time to reply.

pbatard, 2020-05-04:
Reopening this issue as I got a report from someone using Rufus 3.10 getting `[FAIL]  Unable to load driver '\efi\rufus\ntfs_ia32.efi': [3] Unsupported` on an [Acer Iconia model W510p](https://www.acer.com/ac/en/US/content/support-product/4611?b=1&pn=NT.L0SAA.002):

[image]

When using Rufus 3.9 (and therefore an older version of UEFI:NTFS with gnu-efi rather than EDK2 compiled drivers), the error goes away.

I sure wish I had a machine that displays the issue, so that I can figure out exactly why the EDK2 version of the drivers doesn't work, coz having a load failure with the EDK2 drivers makes no sense...

devedse, 2020-05-05:
I still have the laptop.  I can retest this later.  Please keep tagging me and spamming me dip I don't forget.  :)

farid220, 2020-05-06:
Hi guys
I see that you are working on a bug which is apparently a incompatibility with different bootloader headers

Actually this new version of Rufus caused me this error and it says "Unable to start Driver" so I used Rufus 3.9 and it worked just fine.

Now that you are working on this I'll be glad to provide you with result of changes being made in order to solve this problem quickly.

So if you need any tests or stuff to develop the proper bootloader I'm all yours

pbatard, 2020-05-06:
@devedse, @farid220, thanks for offering to help - this is much appreciated.

One thing I am noticing is that the ia32 and aarch64 version of the EDK2 compiled drivers are missing `cpu`, `file-type` and `subsystem`:

[image]

But the gnu-efi version (as well as the x64 and arm version of the EDK2 compiled drivers) have those filled as expected:

[image]

So, yeah, I can certainly imagine why a UEFI 32-bit platform, like the Acer Iconia model W510p might not want to load the drivers at all.

But if that is the issue, then the only people I expect to experience the `[FAIL] Unable to load driver '\efi\rufus\ntfs_ia32.efi': [3] Unsupported` error are those who run a 32-bit UEFI.

@farid220 (and potentially @devedse if you so see an error with Rufus 3.10), can you confirm that your UEFI is 32-bit?

If UEFI:NTFS fails, it's a actually very easy to find that as the line you see on the top, `*** UEFI:NTFS (####) ***` tells you precisely which architecture your UEFI firmware uses between the parentheses.

If I can get confirmation that only ia32 (and aarch64) fail, that would provide some confirmation that I have probably identified the issue.

Oh and I have to point out that, if this is the issue, then some UEFI systems seem to be a lot more tolerant than others, as a Raspberry Pi 4 running a UEFI firmware appears to have no trouble loading the aarch64 driver from UEFI:NTFS, even as it is missing the same data as the ia32 one...

pbatard, 2020-05-07:
Actually, I am able to replicate the issue with QEMU and IA32 OVMF (should have tested that first instead of assuming that I couldn't reproduce this, as was the case with the previous issues):

[image]

On the other hand, the ARM64 QEMU UEFI firmware seems to be absolutely fine about a driver that's missing half the relevant fields, including `entry-point`:

[image]

The other kicker is that, when recompiling the EDK2 drivers using the current version of VS2019, the IA32 PE header fields are properly populated (but not the AARCH64 ones), and I'm also seeing the AARCH64 PE header fields being dropped for the gnu-efi version of the driver...

All in all, it looks like a straight Visual Studio linker bug, that has been existing for a while (all the AARCH64 drivers I compiled with Visual Studio appear to have had this issue regardless of EDK2 or gnu-efi was being used), and that Microsoft appears to have recently introduced and then fixed for the IA32 version. I have validated that we are definitely setting all of the flags that the linker requires to populate the PE header (`/MACHINE`, `/DLL`, `/ENTRY:`, `/SUBSYSTEM`), so if those values aren't filled, it's 100% a linker bug.

Whereas the AARCH64 linker issue doesn't seem that problematic if most UEFI firmwares appear to be able to work around it, I'm really annoyed that the IA32 linker bug was introduced right at the time where I released a new version of the EfiFs drivers, because this then got carried into Rufus. Thanks for breaking my stuff, Microsoft!

devedse, 2020-05-07:
@pbatard ,

I've just setup a clean USB Stick using Rufus 3.10 (from the releases). I then put it in my laptop and made it use `\\...\\efi\rufus\ntfs_x64.efi` but this resulted in just a black screen.

I then downloaded version 1.0 from the website you gave but this resulted in the same issue.

I also tried using the 32 bit version and the exFat version but all of them resulted in just a black screen.

It's been ages since I did anything with this so I'm probably doing something wrong. Any ideas :)?

pbatard, 2020-05-07:
@devedse, thanks for wanting to test this.

The way you should create the USB is make sure that you have the _Advanced drive properties_ showing, because then you will see a _UEFI:NTFS_ option in the _Boot selection_ dropdown, which is what you should use.

Then you just need to place a `bootx64.efi`, `bootia32.efi` in `/efi/boot/` on the NTFS partition, and you're good to go. I'm attaching an archive that contains the files you need to extract on the NTFS partition below (make sure you do extract with the directories: It should contain a `/efi/boot/` folder):

[efi.zip](https://github.com/pbatard/rufus/files/4593431/efi.zip).

Repeat that with different versions of Rufus (especially 3.8 or 3.9 vs 3.10) and it will tell me it there exists other problems besides the IA32 __`load driver`__ error.

pbatard, 2020-05-07:
Okay, I'm only seeing your update now.

Even if it results in the same error code, the `Could not open partition: [3] Unsupported` is a __different issue__, since this time the driver was loaded properly.

What I'd still like to confirm is what you get with the method I highlighted above, using versions 3.10, 3.9 and 3.8 of Rufus.

pbatard, 2020-05-07:
And now I'm starting to see clearer as to what happens for that `[FAIL] Unable to load driver '\efi\rufus\ntfs_ia32.efi': [3] Unsupported` error, which boils down to stupid typos and unlucky false flags.

1. I screwed up the [EfiFs](https://github.com/pbatard/efifs) script that I use to produce a file system driver release, by leaving an unwelcome copy paste that ended up copying/renaming the AARCH64 drivers into the IA32 ones. So the reason why you get `[3] Unsupported` when you load the IA32 drivers is because you are effectively trying to load an AARCH64 driver. Ironically, I had crafted the problematic script to ensure that I wouldn't screw up my uploads, which I renamed and copied over by hand before that...
2. It appears that Pestudio simply does not support AARCH64 (even with the release that happened last month, which I also did test with yesterday), which of course means that there was nothing obvious to indicate that the IA32 drivers we were looking at were really the AARCH64 drivers, since the `cpu` field was not populated... When looking at the AARCH64 drivers with other analysis tools, everything appears in order, including the `subsystem` (which also explains why ARM64 QEMU is fine with).

So at least, the `Unable to load driver` can be explained fully.

Which leaves us with the `Could not open partition: [3] Unsupported` issue, that seems to boil down to using specific hardware as well as whether gnu-efi or EDK2 was used to build the drivers...

devedse, 2020-05-07:
@pbatard , For some reason if I format my USB Stick using UEFI:NTFS option I get the following error (Rufus 3.10):

[image]

This works fine in Rufus 3.9:

[image]

pbatard, 2020-05-07:
Can you please try some more with 3.10. The UEFI:NTFS creation process has actually not changed between 3.9 and 3.10, in that we are creating 2 partitions, and just `dd`'ing 512 KB of data onto the smaller one. In other words, none of the data that was changed for UEFI:NTFS between 3.9 and 3.10 can have an impact on the drive creation, because it belongs in the 512 KB, which Rufus doesn't care one bit about (it just writes it as is) and is treated in the same fashion.

You may want to try <kbd>Atl</kbd>-<kbd>V</kbd> with toggles VDS for formatting, to see if that helps, but there's really nothing that has changed with the manner in which Rufus creates UEFI:NTFS drives between 3.9 and 3.10, so it's probably a transient error, as I have never seen an issue like the one you report on my side. You may also try <kbd>Atl</kbd>-<kbd>Z</kbd> to zero the drive for a little while before you attempt to create the UEFI:NTFS drive...

pbatard, 2020-05-08:
@devedse, any chance for an update again?

I'd like to release a bugfix version of Rufus, so I really would like to have some results from your tests with 3.8, 3.9 and 3.10 when creating UEFI:NTFS drives in the manner I highlighted above:
1. Choose _UEFI:NTFS_ option in the _Boot selection_ dropdown
2. Extract the files from [efi.zip](https://github.com/pbatard/rufus/files/4600496/efi.zip) onto the main NTFS partition (which should be empty apart from the Rufus icon).

Note that I updated the efi.zip archive (in both posts), as it didn't contain quite exactly the files I wanted you to test with, though this wasn't exactly a big deal.

What I am interested is the output you get from your platform when trying to boot these drives. Preferably, I would like you to attempt to boot these drives __without__ going through the `Add Boot Option` utility you seem to be using, as I get a feeling that it might interfere with the boot, and if you do set the proper boot order in the UEFI firmware (or just temporarily disable AHCI/SATA/NVMe there so that only USB should boot), you shouldn't have to do that at all.

Thanks!

devedse, 2020-05-09:
@pbatard , I think somehow my USB Stick got corrupted as I can't format it using windows or DISKPART anymore either.

I now tried with a new USB Stick:

I formatted it using Rufus 3.10, placed the complete EFI directory on the main partition and booted it. This is the result:
https://youtu.be/Ra7u8Y5O_LU

What I don't completely understand now is that we have an EFI\boot folder on both the UEFI disk and the main partition. Is this the thing you want?

What I also tried was simply overwriting the files on the UEFI partition with the ones in the EFI.zip file you sent and that resulted in the following:
https://youtu.be/DRkyNNmGllo

As you can see, in both these scenario's I do need to manually add the option to boot using this specific UEFI configuration, because if I don't it simply won't boot.

devedse, 2020-05-09:
I tried Rufus 3.9 too, but the Bios didn't seem to want to boot from UEFI at all from this stick:
https://youtu.be/BDeR_SRl9-o

I thought this might've been a fluke so I retried the whole process and got the same result:
https://youtu.be/bLqQ1WPUx5s

I then tried Rufus 3.8 which works fine:
https://youtu.be/vzKiqIVPDHE

In the last video I also found out that one boot options gets added if UEFI seems to be correct, which is what I tried booting from in the video of 3.8 at the end. This also worked fine. (But doesn't seem to get automatically selected when just sticking in a USB stick)

Hopefully this all helps you debug the problems :)

pbatard, 2020-05-09:
Thanks for the testing. Let me first address your first post.

> What I don't completely understand now is that we have an EFI\boot folder on both the UEFI disk and the main partition. Is this the thing you want?

Yes, absolutely.

The whole point of UEFI:NTFS is to have a bootloader on a small FAT partition (the one you point to) that loads the NTFS driver and then chain load the actual bootloader on the NTFS partition, so, yes, you definitely want to see a `/efi/boot/bootx64.efi` on __both__ partitions.

What you ultimately want to do is execute the EFI bootloader that resides on the NTFS partition, so that's why if you don't have a `/efi/boot/bootx64.efi`, then there's no point (though of course, if you're just checking for failures of the UEFI:NTFS bootloader then you could theoretically test without the NTFS boot loader, but in that case UEFI:NTFS will always fail even when it should work).

In other words the boot process goes like this:
1. The system (generally) can't boot the NTFS bootloader directly because it can't read NTFS, so it looks for `/efi/boot/bootx64` on the small FAT partition (which it can always read) and execute it. This is the UEFI:NTFS bootloader that is provided by Rufus.
2. The UEFI:NTFS bootloader above attempts to load the missing NTFS driver and, if successful, it hands over the boot process to the `/efi/boot/bootx64.efi` that resides on the NTFS partition. This usually is the boot loader provided by the Windows ISO (i.e. the Windows installer) or, in our case, since we don't want to have to extract a Windows ISO every time, the `/efi/boot/bootx64.efi` that I provided in the `efi.zip` archive above.

> What I also tried was simply overwriting the files on the UEFI partition

Please do __NOT__ do that. You should __NOT__ touch the small UEFI_NTFS partition that already contains the `/Efi` and `/Rufus` folders. If you do that, then you will effectively erase the UEFI:NTFS bootloader with something that is not UEFI:NTFS (the bootloaders from `efi.zip` simply display a message, they are not "test" versions of UEFI:NTFS that you should replace the existing UEFI:NTFS with).

> As you can see, in both these scenario's I do need to manually add the option to boot using this specific UEFI configuration, because if I don't it simply won't boot.

And now I realize that your boot selection window is not a Microsoft Windows application, but a UEFI menu. So yes, that's what you should use to select the device you want to boot from (though I'm a bit puzzled that Dell forces you to point to a specific file instead of just automatically looking for a boot partition, which is how UEFI is set to proceed... count on Dell to make it a lot less user intuitive than needs to be).

pbatard, 2020-05-09:
I have a question though: What happens if you simply select `Stick2` without explicitly clicking `Add boot option` and picking a file? Coz, once again, UEFI is designed to __automatically__ locate a and boot a `/efi/boot/bootx64.efi` boot loader on any partition that it can read, so the one from the FAT partition should be picked automatically by a UEFI compliant system simply by telling it that you want to boot the USB drive in UEFI mode.

pbatard, 2020-05-09:
Okay, now to comment on your videos:
* https://youtu.be/iRyV4h4DTRQ
  Yes, that is the correct procedure (PS: Wow, you're uploading 4K videos - nice!! 😄)
* https://youtu.be/Ra7u8Y5O_LU
    This looks like the gnu-efi vs EDK2 compiled issue that I am after. The NTFS driver that's included in Rufus 3.10 is the EDK2 version, which some folks report is causing trouble, as opposed to the gnu-efi compiled version.
* https://youtu.be/DRkyNNmGllo
  No, that won't work. Forget about doing this, coz that's not what I want you to do.

* https://youtu.be/BDeR_SRl9-o
  I really can't see how that can happen unless there was an issue during the drive creation and you ended up with a non UEFI:NTFS drive. Can you please re-test this, and post the complete log from Rufus from how you created the drive. I really can't see how the Dell UEFI firmware would skip the FAT partition, when we're using the exact same as 3.8 and 3.10 to create it.
* https://youtu.be/bLqQ1WPUx5s
  Again, a Rufus log will help.
* https://youtu.be/vzKiqIVPDHE
  Yes, this is the expected result, and the one you'd get from a system that has trouble with the EDK2 version of the NTFS file system driver but not with the EDK2 version.

If possible, I'd be very curious about what happens if you run the UEFI Shell and try to load the NTFS driver from there, and whether you'd see the same errors. Let me prepare a procedure to guide you through that.

[27 more replies]

commit 7463fa6d (2020-05-26): [efi] update the UEFI:NTFS image

* Now use version 1.6 of the EfiFs drivers that enables firmwares that
  don't provide EFI_DEVICE_PATH_TO_TEXT_PROTOCOL to mount NTFS partitions
  regardless.
* Also use the latest version of UEFI:NTFS that displays additional info
  about the system.
* Closes #1213.
* Also update version to Rufus-next.
closed 2020-05-26 by pbatard