pbatard/rufus #1214: ERROR: Root device mounted successfully, but sbin/init does not exist.

johnsonlukose, 2018-09-23:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Trying to make bootable USB for manjaro-xfce-17.1.11-stable-x86_64. 
Trying to make bootable USB for manjaro-xfce-17.1.12-stable-x86_64. 

It fails for both iso(s)...

Tried following parameters on both Rufus 3.1 & Rufus 3.3 
Partition scheme: MBR 
Target system: BIOS or UEFI 
File system: FAT32 (Default) 
Cluster size: 4096 bytes (Default) 

Partition scheme: GPT 
Target system: UEFI (non CSM) 
File system: FAT32 (Default) 
Cluster size: 4096 bytes (Default) 

Log
Starting version 239
/dev/sda2: recovering journal
/dev/sda2: clean, 1975/1310720 files, 129714/5242880 blocks
ERROR: Root device mounted successfully, but sbin/init does not exist.

Bailing out, you are on your own. Good luck. 

sh: can't access tty: job control turned off 

[rootfs ]# _ 

```
<FULL LOG>
```

pbatard, 2018-09-23:
Yeah, the Manjaro people don't seem to care about supporting FAT or NTFS file systems. See [Manjaro issue 86311](https://bugs.manjaro.org/#ticket/zoom/17934).

I have reported that problem almost 4 months ago, and they don't seem to want to do anything about it. Until they do, I am not going to bother trying to support Manjaro, because they really need to stop putting all their eggs in the ISOHybrid basket.

Closing as _"Needs to seem some progress from the Manjaro distro maintainers first"_.

PS: Remember that big prompt that asked you whether you wanted to write the image is ISO or DD mode, and that __specifically__ told you to try DD mode if you had an issue in ISO mode. You should try that.

closed 2018-09-23 by pbatard

johnsonlukose, 2018-09-23:
Thank you for your reply. 

I had gone back to the "old method" of DVD using CDBurnerXP to make an ISO DVD from manjaro-xfce-17.1.12-stable-x86_64 & I got the SAME RESULT... 

I have given-up and installed MX Linux...

thanks.