pbatard/rufus #1215: Two controls in Rufus main windows lacks proper labels for assistive technologies.

lukaszgo1, 2018-09-24:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
Issue description
-----------------
The button for creating a vhd from connected usb disk and the button for calculating checksum for a currently selected image aren't labeled. It makes it harder for a blind user who relies on a screen reader to deduce their functions. Other buttons Such as ok. or log are labelled,, sou it definitely is possible to do.

pbatard, 2018-09-24:
Thanks for the report. I'll see what I can do.

But can you please confirm that you really get a label for the log button? 

I'm a bit surprised to hear this as the button is part of a 4 button toolbar that was tricky to set up and that can't be navigated through tabbing because, due to Windows' weird accessibility limitations for toolbars, one must first tab into the toolbar and then use the arrows, instead of Tab, to navigate to each individual button there, including the log button which is the last one. My expectation was that this would make it quite difficult to people using screen-readers to navigate to the log button, but I was somewhat hopeful that most of them would be aware of the quirk or requiring directional arrows instead of Tab for navigation there. 

So if you can confirm that you did indeed navigate to the log toolbar button and that this button was properly labelled, it should help me figure out what I can do to label the other buttons.

I can't promise anything though, as these toolbars (all of these buttons, even the individual ones that are missing labels, reside on a toolbar) have been a major pain, and I am having other issues with the 4-button toolbar, regarding tooltips, that make me a bit weary of trying to make the other toolbars behave like the 4-button one.

lukaszgo1, 2018-09-24:
@pbatard wrote
> Thanks for the report. I'll see what I can do.
> 
> But can you please confirm that you really get a label for the log button?

The log button definitely has proper label.

> I'm a bit surprised to hear this as the button is part of a 4 button toolbar that was tricky to set up and that can't be navigated through tabbing because, due to Windows' weird accessibility [...]

Frankly use of toolbars in such way isn't often seen, but i believe, that Rufus is a tool used by more advanced users, so it isn't a big deal.

> So if you can confirm that you did indeed navigate to the log toolbar button and that this button was properly labelled, it should help me figure out what I can do to label the other buttons.
>
> I [...]

No problem. It isn't a big issue, because you can route mouse to any of these buttons, and check what it does by reading the tooltip.

pbatard, 2018-09-24:
Thanks for the update. Glad to hear that toolbar button accessibility might not be as big a deal as I feared it would. I'll see what I can do for the labels.

pbatard, 2018-10-05:
Issue should be fixed now. I tested the fix with nvda and it will be included in Rufus 3.4.

commit 19d9bdbe (2018-10-05): [ui] fix missing description of some buttons for screen readers

* Closes #1215
closed 2018-10-05 by pbatard