pbatard/rufus #1219: Rufus error,Unable to create file: [0x00000005] Access is denied.

Bcarmona78, 2018-09-29:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ x] I also tried one or more of the following:
  - [ x] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<I recently downloaded a Windows 10 ISO file from the official website using a separate computer to prepare a USB drive for booting Windows 10 on a new computer I built. I discovered Rufus as a good option for doing so, downloaded the newest version and setup everything as I should. I began the process and a few seconds in I got the following error. I looked at some other posts about similar issues but couldn't find a definite solution. Thank you for your help whether you can help solve the issue or not.    >

Log
---
```
<Rufus version: 3.3.1400
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'SanDisk Cruzer Glide USB Device' (0781:5575)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1869, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x004474F6
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.3 GB (15375269888 bytes)
  Start Sector: 2048, Boot: Yes
Checking for Rufus updates...
Updates: Unable to connect to the internet
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Size: 4.4 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_1803_English_x64.iso (4.4 GB)

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{ff00d77e-dbb9-11e8-8f45-c45444b507db}\
Disabling file indexing...
Waiting for access on \\?\Volume{ff00d77e-dbb9-11e8-8f45-c45444b507db} [\Device\HarddiskVolume8]...
Opened \\?\Volume{ff00d77e-dbb9-11e8-8f45-c45444b507db} for exclusive write access
Writing partition boot record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted \\?\Volume{ff00d77e-dbb9-11e8-8f45-c45444b507db}\ on D:
Copying ISO files...
Extracting files...
Image is an UDF image
Extracting: D:\autorun.inf (128 bytes)
Extracting: D:\boot\bcd (16 KB)
Extracting: D:\boot\boot.sdi (3 MB)
Extracting: D:\boot\bootfix.bin (1 KB)
Extracting: D:\boot\bootsect.exe (104.9 KB)
  Unable to create file: [0x00000005] Access is denied.

Found USB 2.0 device 'SanDisk Cruzer Glide USB Device' (0781:5575)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1869, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00467E05
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.3 GB (15375269888 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume GUID \\?\Volume{ff00d77e-dbb9-11e8-8f45-c45444b507db}\
Disabling file indexing...
Waiting for access on \\?\Volume{ff00d77e-dbb9-11e8-8f45-c45444b507db} [\Device\HarddiskVolume8]...
Opened \\?\Volume{ff00d77e-dbb9-11e8-8f45-c45444b507db} for exclusive write access
Writing partition boot record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted \\?\Volume{ff00d77e-dbb9-11e8-8f45-c45444b507db}\ on D:
Copying ISO files...
Extracting files...
Image is an UDF image
Extracting: D:\autorun.inf (128 bytes)
Extracting: D:\boot\bcd (16 KB)
Extracting: D:\boot\boot.sdi (3 MB)
Extracting: D:\boot\bootfix.bin (1 KB)
Extracting: D:\boot\bootsect.exe (104.9 KB)
  Unable to create file: [0x00000005] Access is denied.

Found USB 2.0 device 'SanDisk Cruzer Glide USB Device' (0781:5575)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1869, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00478013
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.3 GB (15375269888 bytes)
  Start Sector: 2048, Boot: Yes
>
```

pbatard, 2018-09-29:
https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied

Bcarmona78, 2018-09-29:
Thank you for sharing that forum, although it did not solve my problem. I'm aware that there are many potential causes for my issue and I will continue to investigate. Do you have any suggestions or ideas for the cause of access being denied to my USB, thanks again, Brandon.

pbatard, 2018-09-30:
Security software would be a good culprit. But it can be anything, really, and it's up to you to figure out what is preventing access.

pbatard, 2018-10-01:
I appreciate the tip, but I'd very much prefer knowing __precisely__ what is causing the access denied, instead of doing the equivalent of shooting in the dark and hope for the best.

Hate to pull rank here, but more than 30 years spent developing software have taught me, repeatedly, that fixing issues you don't precisely understand the underlying causes of can and does create much worse problems than not fixing the issue at all, especially if the issue is very limited in scope (which this one is, as it only appears to affect one person in the tens of millions who use Rufus regularly as I believe I am monitoring enough input channels to be able to assert that this is everything but a widespread problem).

Therefore, unless I can replicate the issue, and understand what causes it, or OP can provide information as to what environmental specifics of their system is likely to be the culprit (which I can then try to corroborate), I'm not going to apply any change, no matter how innocuous it may look, just to __try__ to address what is so far a single person's issue.

Besides, my experience is that there do exist some overzealous security solutions out there that try to "lock" people out of modifying boot files. If this is what is preventing access, then adding the flag won't do much anyway.

TL;DR I don't want to "fix" an issue I don't understand, especially if it only seems to affect a single person out of the tens of millions of regular Rufus users.

Mattiwatti, 2018-10-01:
Understood, that's perfectly fine. It was only a suggestion, no '**this is what you should do**' intended :wink: I suggested it only because I have read the piece of kernel source code that deals with this flag, and its effects are pretty clear and precisely match MSDN's description. I've also fixed seemingly 'random' file permission errors like this one by appending this flag several times (though admittedly it's a rare occurence).

As for "security" products causing this: I totally agree, this is always a possibility that is outside the program's control. And no one hates such software more than I do. But, believe it or not, there **are** some vendors who do read the docs and respect the semantics of this flag! (that's the second rare occurrence.)