pbatard/rufus #1220: rufus.com at ARM version

hooddy, 2018-10-01:
Rufus.exe contain rufus.com. To run rufus on WinRT i use sign tool. rufus.exe runs well excepts rufus.com. So maybe in arm version rufus.com is not needed and must be exclude?

pbatard, 2018-10-01:
`rufus.com` is only needed when launching from commandline, which is not supported on ARM. Therefore there's no point actively excluding it.

closed 2018-10-01 by pbatard

pbatard, 2018-10-01:
By the way, I am providing official signed versions of Rufus for ARM and ARM64 [here](https://rufus.akeo.ie/testing/), so you shouldn't really have to sign anything.

hooddy, 2018-10-01:
Anytime i run rufus in winrt it try to run rufus.com and i get errorcregarding this issue coz under winrt it cent be run. Although rufis.exe runs well. But error occurs anytime

hooddy, 2018-10-01:
I download it right from this link. What i cen do to avoid error regards rufus.com?

hooddy, 2018-10-01:
Btw rufus for win desktop doesnt have rufus.com so why it exists in rufus for winrt? I really dont undastood.

pbatard, 2018-10-01:
Other people using WinRT seemed to have no trouble running it.

Again, `rufus.com` only launches if Rufus detects that it is being launched from the commandline. Therefore, there must be something in your environment that triggers this. You will have to figure out what that is. Are you using the native Windows interface to launch Rufus?

pbatard, 2018-10-01:
> Btw rufus for win desktop doesnt have rufus.com

It does. Your assertion is wrong.

hooddy, 2018-10-01:
Please reopen issue till fixed it.

hooddy, 2018-10-01:
Well ok. I dont care anymoar.

pbatard, 2018-10-01:
> About rufus for deaktop. I open appx amd within it i open rufus.exe in hex view and didnt find any rufus.com trace.

The reason for that is because the `.appx` and x86 desktop versions are __compressed__ using LZMA. So any string like `rufus.com` will be compressed and you won't be able to find it if you look for it.

__However__ if you decompress these files, you will find that the `rufus.com` string (which is really mentioned only as part of the commandline options for what we call the "hogger", see [here](https://github.com/pbatard/rufus/tree/master/res/hogger) if you are technically inclined and want to know more), is very much present in these executables.

Furthermore, I can very much point to the `rufus.com` symbols you'll find in the __uncompressed__ executable, which come from [here](https://github.com/pbatard/rufus/blob/204908f8e00053d5b007e25de7dab1d3f926b8de/src/rufus.c#L54) and [here](https://github.com/pbatard/rufus/blob/204908f8e00053d5b007e25de7dab1d3f926b8de/src/rufus.c#L2664) (and if you understand C, you'll see that there's nothing in the code that says _"only use these for the ARM version"_, and therefore, that, as I indicated, you will find `rufus.com` in __all__ versions of Rufus.

> And it smaller about 2md then rufus for arm.

Yes. That's the whole point of compression. If there was executable compression like UPX for ARM, I would use it. But there isn't (yet, at least for PE executables), so the ARM `.exe`'s are uncompressed, which means they are bigger and you can see uncompressed strings like `rufus.com` with an hex editor.

> If i open rufus.exe for winrt in hex view i find rufus.com trace in it. So i just suppose that desktop version have no com file in it.

No, once again your assertion is wrong. There are only so many way I can indicate that `rufus.com`, a.k.a. the commandline hogger, which Rufus needs to use to work around the long standing problem of Windows not allowing an application to properly work both for commandline  as well as GUI, is present in every version of Rufus, regardless of the architecture, and that the reason you see it appear on your system is because, _somehow_, Rufus thinks it was launched from the commandline.

But considering that I have reports from other people running Rufus/ARM on WinRT who don't seem to run into this problem, I have to assert that your issue has to be linked to your specific environment.

hooddy, 2018-10-01:
Ok ok. Whos care about my issue yeah. Maybe it coz i run it from far manager. Ok. But anyway for what the reason put com file into unsupported environment like winrt or arm64 i really dont undastood. Is it so hard to find compile exe with no com inside? Just change some code. Arm64 i suppose have this issue.

pbatard, 2018-10-01:
> Maybe it coz i run it from far manager.

It is! Far manager is using the command line application behind the scenes, which makes it a complete pain in the ass to support.

> But anyway for what the reason put com file into unsupported environment like winrt or arm64 i really dont undastood. 

Precisely __because__ far manager is part of the reason we have `rufus.com` (i.e. the commandline hogger). See the readme at https://github.com/pbatard/rufus/tree/master/res/hogger, and especially the following section:

> 5. To compensate for this, you might try to be (somewhat) clever, through the simulating an <Enter> keypress when your app exit, but lo and behold, soon enough you start receiving complaints left and right from Far Manager [3] users, due to the fact that this application uses the <Enter> keypress as an "I want to launch the currently selected app" event.

> Just change some code.

Just don't use Far Manager.

I spent enough time trying to help Far Manager users, and have since found it's simply not worth it. SO I will respectfully that you either don't launch Rufus with Far Manager, or use a different application, as __I will not provide any further input or support on that issue__.

hooddy, 2018-10-01:
Its a lil strange to advice to dont use console manager while your app have console code. Well ok. Let it go. Cheers)