pbatard/rufus #1222: "Cannot Connect to Network"

TheSovis, 2018-10-02:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the "Log" button in Rufus (last small blue button before _START_) and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

TheSovis, 2018-10-02:
Hello,

In my organization I have wireless usb adapters connected to Rufus routers network. Everything works perfectly fine, except of when I'm trying to install TP-Link TL-WN821N model adapters. 
I can see all available networks, but after typing username and password, the message "Cannot connect to this network appears. 
Drivers are good as well as users have administrator privileges.     

I would be very thankful if you could help me in solving this issue.

pbatard, 2018-10-02:
> Rufus routers network

This is an issue tracker for the Rufus Windows __software__, that is used to create bootable USB devices.

Since "Rufus" is a relatively common name, I have no doubt that __completely unrelated__ hardware may be called "rufus" too. Unfortunately, as I hope you can understand, that hardware has absolutely nothing to do with our software. As such I will ask you to please contact the proper support channel for your hardware, which this isn't.

closed 2018-10-02 by pbatard